FreeOpcUa/python-opcua #1281: Source Timestamp

ksrangini05, 2021-03-05:
I am  getting the data from a DB  and assigned with the source timestamp. When it is connected with UA expert tool, this assigned source timestamp is coming . When the data is updating, the client tool is taking the local system time, instead of assigned source timestamp. Please find the code for reference. Anybody faced this issue and solved it ?
datavalue = ua.DataValue(inpoints[0]['value']
datavalue.SourceTimestamp = datetime.fromtimestamp(inpoints[0]['time'])
server.get_node(ua.NodeId.from_string('ns='+str(idx)+';s='+str(j['requested_nodeid']))).set_attribute(ua.AttributeIds.Value, datavalue)

ksrangini05, 2021-03-16:
[image]
PFA the UA expert screenshot for the reference

AndreasHeine, 2021-03-16:
thats normal the attributes tab only got updated if you click/browse the node in the nodetree!

ksrangini05, 2021-03-16:
But the source timestamp of the value is different in attribute and in data access view.

AndreasHeine, 2021-03-16:
Sorry cant confirm any issue...

[image]
```python

import datetime
import time
import logging
from opcua import Server, ua

logging.basicConfig(level=logging.CRITICAL)
logger = logging.getLogger("opcua")
logger.setLevel(logging.CRITICAL)

old = datetime.datetime.utcnow()
val = 500.0
print(old, "-->", val)
time.sleep(2)

server = Server()
idx = server.register_namespace("http://test.de/UA")
obj = server.get_objects_node()
myvar1 = obj.add_variable(idx, "MyVar1", 0, ua.VariantType.Float)
server.start()

dv = ua.DataValue(
    ua.Variant(val, ua.VariantType.Float), 
    ua.StatusCode(ua.StatusCodes.Good),
    sourceTimestamp=old,
    serverTimestamp=datetime.datetime.utcnow()
    )
myvar1.set_value(dv)
```

closed 2021-03-16 by AndreasHeine