FreeOpcUa/python-opcua #1282: call_method passing ExtensionObject argument does not support Bit attribute

apisapiaDL, 2021-03-08:
Hello!
I am trying to develop an OPC UA client using “Free OPC UA” in order to test an OPC UA server with an AutoID Companion Specification (It defines some object types with methods and Data Types).

One of these methods takes as an input a ScanSettings parameter defined in the following way in the bsd file:

[image]

As you can see, the first two fields define a bitmask that figures out which optional fields are specified in the instance of the structure (in this case just LocationType is optional).
When I call such method in the following way, I get this error:

[image]

and the following disassembly code:

[image]

So, I am wondering if this kind of structure with optional fields are supported by the library and how could be fixed?

I've also tried passing the custom DataType "AntennaNameIdPair" which does not defines optional fields, and the call method seems to work fine.

Python-Version: 3.8
python-opcua Version): 0.98.13

AndreasHeine, 2021-03-08:
please provide a simplyfied example code which reproduces the error!

apisapiaDL, 2021-03-08:
Hi!
Here you can find my OPC UA Server
[OPC Server.zip](https://github.com/FreeOpcUa/python-opcua/files/6101890/OPC.Server.zip)

and here my client Visual Studio solution, the script is pythonOPCUA.py
[pythonOPCUA.zip](https://github.com/FreeOpcUa/python-opcua/files/6101903/pythonOPCUA.zip)

Thank you for your support!
Amedeo

zerox1212, 2021-03-08:
I doubt anyone here is going to grab your entire project and troubleshoot it for you. You need to make a simplified example showing how the `Bit` attribute isn't working...

apisapiaDL, 2021-03-08:
> I doubt anyone here is going to grab your entire project and troubleshoot it for you. You need to make a simplified example showing how the `Bit` attribute isn't working...

The first zip contains just an executable Server with AutoID Companion.
The second zip is a very simple client with just 43 lines... ([Client script.zip](https://github.com/FreeOpcUa/python-opcua/files/6102568/Client.script.zip))

Amedeo

dlauser, 2021-03-08:
I guess, what apisapiaDL is meaning is:
- he instanced the autoid companion on the server (as defined by opcua nodeset)
- the client loads the types from theserver and builds a types dictionary
- now he needs to call a method on it but _arguments_ is a datatype that uses opc:Bit option fields that
 FreeOpcua client seems  not being able to support, in fact client library seems to work fine with other ExtensionObjects.

So, the question might be: 
does anybody know whether this library has any issues with Bit attributes in ExtensionObjects or simply there is a specific way to handle them on the client side?

AndreasHeine, 2021-03-08:
i will never ever run a exe from someone i dont know!

i am thinking what the diference is "opc:bit" <-> "opc:boolean" 
in my opinion "opc:boolean"

dlauser, 2021-03-08:
> l never ever run a exe from someone i dont know!
neither would I!

oroulet, 2021-03-08:
Can you test with opcua-asyncio, sync API? I think optional fields are missing at some places in python-opcua.
But this is the old format and it is not supported. try to call load_data_type_definition() and see if your server support newer strcut definition

deeTEEcee, 2022-01-31:
I believe I'm having a similar issue. You can try to load up an xml with a type dictionary that includes something like 
```
<opc:TypeDictionary
      <opc:StructuredType BaseType="ua:ExtensionObject" Name="ProcessValueType">
            <opc:Field TypeName="opc:Bit" Name="cavityIdSpecified" />
            <opc:Field TypeName="opc:Bit" Name="descriptionSpecified" />
            <opc:Field Length="30" TypeName="opc:Bit" Name="Reserved1" />
            <opc:Field TypeName="opc:CharArray" Name="name" />
            <opc:Field TypeName="opc:Double" Name="value" />
            <opc:Field TypeName="opc:UInt32" Name="assignment" />
            <opc:Field TypeName="opc:UInt32" Name="source" />
            <opc:Field SwitchField="cavityIdSpecified" TypeName="opc:UInt32" Name="cavityId" />
            <opc:Field TypeName="opc:CharArray" Name="id" />
            <opc:Field SwitchField="descriptionSpecified" TypeName="opc:CharArray" Name="description" />
         </opc:StructuredType>
```

Reconstructing that type with the Bit field will lead to `AttributeError: module 'opcua.ua' has no attribute 'Bit'`

[Bit implementation](https://github.com/FreeOpcUa/python-opcua/blob/master/schemas/generate_model.py#L33-L41) here seems somehow relevant but I think I have to read a lot more source code and figure out how to have that included as part of the code generation. Is it some legacy code that we threw away in the past?

deeTEEcee, 2022-01-31:
I can't use the newer async library either because that seems to require DataTypeDefinition objects which exist in the server but im working from the client side and have no control over what is on the server. If I use that lib, I just get messages like this one: `DEBUG:asyncua.common.structures104:ns=1;g=c2089763-b599-482c-9664-e7fa2fa3b654 has no DataTypeDefinition attribute`

oroulet, 2022-01-31:
The problem i that optional fields are not implemented in the parsing of xml form the server. Here is the code: https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/common/structures.py#L137

one should create that "Encoding" dataclass menember if a bit is decclared. Like this: https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/common/structures.py#L137. than make sure that the field in question is declared as optional later while paring all fields

yo should implement something similar to

deeTEEcee, 2022-02-08:
A bit confused why when i made a change like this:
```
def extensionobject_from_binary(data):
    """
    Convert binary-coded ExtensionObject to a Python object.
    Returns an object, or None if TypeId is zero
    """
    typeid = nodeid_from_binary(data)
    Encoding = ord(data.read(1))
    body = None
    length = Primitives.Int32.unpack(data)
    if Encoding & (1 << 0):
        # length used to be here
```

I end up getting an unrelated error at the very early stage before this method even gets called. The error comes from OpenSecureChannelRequest -> Primitives.Int32.unpack call -> NotEnoughData error because trying to pull 4 bytes when only 1 exists.

deeTEEcee, 2022-02-16:
@oroulet thanks, I believe I've resolved this issue so I'll make a PR for it sometime soon.

oroulet, 2022-02-16:
I can merge anything. But if the bugs exist for both then it would be very nice with a bug fix in asyncua

deeTEEcee, 2022-02-16:
ya i can add it in for both libraries.

deeTEEcee, 2022-04-01:
[python-opcua PR to support custom struct w/ optional fields](https://github.com/FreeOpcUa/python-opcua/pull/1451)

deeTEEcee, 2022-05-23:
it looks like i got some assistance as someone else helped me make the PR for opcua-asyncio: https://github.com/FreeOpcUa/opcua-asyncio/pull/900