FreeOpcUa/python-opcua #1285: remove debug print...

ulda, 2021-03-12:
Somebody left a debug print in the code.

diff --git a/remi/gui.py b/remi/gui.py
index c33b5fd..4c4dba3 100644
--- a/remi/gui.py
+++ b/remi/gui.py
@@ -2037,7 +2037,7 @@ class TabBox(Container):
     @decorate_set_on_listener("(self, emitter, key)")
     @decorate_event
     def on_tab_selection(self, emitter, key):
-        print(str(key))
+        #print(str(key))
         for k in self.children.keys():
             w = self.children[k]
             if w is self.container_tab_titles:

AndreasHeine, 2021-03-13:
please copy and post the permalink!

ulda, 2021-03-14:
I was not able to format the patch correctly within  the mail. I'l attache it here.
[remi-remove-print.patch.gz](https://github.com/FreeOpcUa/python-opcua/files/6136336/remi-remove-print.patch.gz)

AndreasHeine, 2021-03-14:
Make a Pullrequest or tell us the file and line!

ulda, 2021-03-14:
Oh, I thought the patch woud be sufficient.
But look at remi/gui.py line 2040

AndreasHeine, 2021-03-14:
are you sure you are in the right repository?

ulda, 2021-03-14:
O Sh...  <peinlich>  - Sorry.

closed 2021-03-14 by ulda