FreeOpcUa/python-opcua #1289: Attributes of DataTypes are not correctly handled by an opcua server

thunderbug1, 2021-03-18:
I did tests with custom data structures with this [example](https://github.com/FreeOpcUa/opcua-asyncio/blob/3b30f4332e842407a3afbc67ab3af9d64cb401b3/examples/server-create-custom-structures.py) but when I open the server with the [opcua-client ](https://github.com/FreeOpcUa/opcua-client-gui) and navigate to  Root/Types/DataTypes/BaseDataType/Structure then it does not show how the attributes of any of the types. Shouldn't there be a detailed definition of what is contained? I also opened a connection to an opcua server on an S7 PLC and there the opcua-client does show attributes for the types, so it must be an issue either with the example or the library.

[image]

oroulet, 2021-03-18:
try the opcua-asyncio server.  if there is a big there I can fix it

thunderbug1, 2021-03-18:
oh, quick response :)

where is that server?

oroulet, 2021-03-18:
https://github.com/FreeOpcUa/opcua-asyncio

oroulet, 2021-03-18:
it replaces python-opcua

thunderbug1, 2021-03-18:
Doesn't the [example](https://github.com/FreeOpcUa/opcua-asyncio/blob/3b30f4332e842407a3afbc67ab3af9d64cb401b3/examples/server-create-custom-structures.py) use opcua-asyncio?

I ran the example server on my laptop and connected to it with the opcua-client.

Sry for my confusion.

swamper123, 2021-03-18:
I think you are in the wrong repo. This is the old- sync one. I will switch it to opcua-asyncio.