FreeOpcUa/python-opcua #1290: data change callback needs status code, timestamp, etc.

GrantEdwards, 2021-03-18:
### Please read, before you post!

This is a _**BUG REPORT for issues in the existing code**_. 

If you have general questions, code handling problems or ideas, please use the:

Discussionsboard: https://github.com/FreeOpcUa/python-opcua/discussions <br/> or <br />
Gitter-Channel: https://gitter.im/FreeOpcUa/python-opcua

---

To save some time, please provide us following informations, if possible:

**Describe the bug** <br />
A clear and concise description of what the bug is.

**To Reproduce**<br />
Steps to reproduce the behavior incl code. <br />
If you post code, please use the "insert Code" (which is the "< >" symbol) format for a better readability.

**Expected behavior**<br />
A clear and concise description of what you expected to happen.

**Screenshots**<br />
If applicable, add screenshots to help explain your problem. <br />
You can post Screenshots directly to github, please don't use 3rd Party webpages

**Version**<br />
Python-Version:<br />
python-opcua Version (e.g. master branch, 0.9):

AndreasHeine, 2021-03-18:
https://github.com/FreeOpcUa/python-opcua/blob/bf734dfc00e96b3ba1c184c5f511589fa183166b/opcua/common/subscription.py#L153

GrantEdwards, 2021-03-18:
I'm sorry, I didn't intend to create that issue. When I started looking at the code to suggest a change, I realized what you just pointed out: that what I wanted was event_data.monitored_item.Value. Somehow, when I tried to "discard" the issue I submitted it instead.

closed 2021-03-18 by GrantEdwards