FreeOpcUa/python-opcua #1329: Handling all possible opcua exceptions using freeopcua

rajnishnationfirst, 2021-05-30:
Hi team ,

 I want to handle the  below exceptions using **python opcua** exceptions  like below .

**1) BadSessionIdInvalid 
2) BadSessionNotActivated
3)BadTooManyOperations
4) BadSessionIdInvalid**

suppose i get the exceptions as :

**a) opcua.ua.uaerrors._auto.BadTooManyOperations: The request could not be processed because it specified too many operations.(BadTooManyOperations)**

AND 

**b) opcua.ua.uaerrors._auto.BadSessionIdInvalid: The session id is not valid.(BadSessionIdInvalid)**

I tried to use the below import statement .

1) **import stament of ua eerors**
2) **import opcua.ua.uaerrors._auto.BadTooManyOperations**

But i was not able to import the below exception classes to handle in the except block like.

```
try:
    ..... code

except (BadSessionIdInvalid ,BadSessionNotActivated,BadTooManyOperations,BadSessionIdInvalid) as errors:
     ....code 
```

Need help for the handling the exceptions . Request anyone to help with your replies for below questions.

1) Correctt import statatemt of opcua exceptions ( uaerros )

2) Any base class to handle all exceptions of opcua.

It will be great help of me if any one can assist the same .

Rajnish Vishwakarma

AndreasHeine, 2021-05-30:
not an issue!

closed 2021-05-30 by AndreasHeine

oroulet, 2021-05-31:
also if someone comes here the correct syntax is.
```
try: 
   mys ua code
except ua.UaError:
    do something
```

rajnishnationfirst, 2022-09-08:
Hi Oroulet ,

Thanks a lot for your kind support for me as i have been very new OPC UA protocol.

If i have to receive all possible exceptions for a client-minimal.py  example of free-opcua python , is the below code correct to catch all the possible exceptions occurring mentioned above in my first post to this discussion : 

**
1) BadSessionIdInvalid 
2) BadSessionNotActivated
3) BadTooManyOperations
4) BadSessionIdInvalid
**

Is the below code correct for all exceptions mentioned above ? 
 
```
class SubHandler():

    try: 
         def datachange_notification(self, node, val, data):
             print(datetime.now(), "- Data Change - ", val)

        def event_notification(self, event):
            pass
  except ua.UaError:
        do something like reconnect or resuscribe or etc
```

Thanks

Rajnish Vishwakarma

swamper123, 2022-09-08:
If you are excepting `UaError` you may get one of those:

https://github.com/FreeOpcUa/python-opcua/blob/858274721766907148b527916e97446dbbf6af6d/opcua/ua/uaerrors/_auto.py

Since stuff can change, you can't guarantee that just those 4 will occur.

swamper123, 2022-09-08:
And I guess you are aware of #1475  ?!

rajnishnationfirst, 2022-09-08:
Thanks @swamper123 for your reply.
yes agreed . Could you pls let me know whether the below code for handling all possible exceptions is correct ?

Code is as follows : 

**Is the below code correct for handling all exceptions in client-minimal example mentioned above** ?

```
class SubHandler():

    try: 
          def datachange_notification(self, node, val, data):
               print(datetime.now(), "- Data Change - ", val)
  
          def event_notification(self, event):
              pass

    except ua.UaError:
          do something like reconnect or resuscribe or etc
```

Thanks

Rajnish Vishwakarma

swamper123, 2022-09-08:
The code you are posting seems to handle `UaErrors`.
Follow the link in my first answer to see, what is expected what a server may return.

I can't say if this list is uptodate, since this library is deprecated and not sure if even the nodeset from asyncua is uptodate atm.
It is possible that more errors may be possible.