FreeOpcUa/python-opcua #1342: OPCUA Client

FabioAlmeida25, 2021-06-24:
Hello guys can anyone help me please ? 
[image]
[image]

HaythemLtifi, 2021-07-01:
its time out . are you sure you have a server.py running? 
i thing you are trying to connect to none existing server and that is throwing the connecting time out flag.
hope this help

swamper123, 2021-07-01:
Please use the Bug-Report Template (the text which is already written, when you open a new issue).
Otherwise we don't know what you try, what you have and what shall be done etc.

These screenshots just tells us: The Server doesn't respond to the client and timed out.