FreeOpcUa/python-opcua #1345: demo server code is broken

HaythemLtifi, 2021-06-30:
Hello , not sure what is the problem but i think my library is broken .the minimal python sever code does no work. it compiles but do nothing.
I think the node.add_object is the broken one. can someone confirm?
using windows 10 and python 3.7.9
[image]

AndreasHeine, 2021-06-30:
can you explain the issue?

swamper123, 2021-06-30:
Please use the "Bug Report" Template next time.
Beside your choice of var names (which confuse me a bit), what exactly doesn't work? Is an issue thrown?

HaythemLtifi, 2021-06-30:
node.add_object should return a object right? in my case it does not . param type is any. 
if you notice the add_variable color is not yellow. 
its as if node.add_object does not work

HaythemLtifi, 2021-06-30:
> 
> 
> Please use the "Bug Report" Template next time.
> Beside your choice of var names (which confuse me a bit), what exactly doesn't work? Is an issue thrown?

yes i will thanks.
i think the node.add_object method has an issue . it does not return an object type

HaythemLtifi, 2021-06-30:
same thing with the demo code
[image]

HaythemLtifi, 2021-06-30:
using this tool i found out that the server works , but the discolored throw me away..... sorry guys i am new and have been following a youtube video. 
but why the add_variable is not colored? in the video its colored . its like param is not an object. but it should be right?
[image]

AndreasHeine, 2021-06-30:
> node.add_object should return a object right? in my case it does not . param type is any. 
> if you notice the add_variable color is not yellow. 
> its as if node.add_object does not work

it should return a nodeclass instance!

AndreasHeine, 2021-06-30:
> same thing with the demo code
> [image]
> 

add_variable() is a method of the nodeclass instance

AndreasHeine, 2021-06-30:
https://github.com/AndreasHeine/python-opcua/blob/bf9dc941acf4da53ebe7505108d885d9e77cee75/opcua/common/node.py#L31

HaythemLtifi, 2021-06-30:
param type is any . IS it correct? shouldn't it be node?

[image]

AndreasHeine, 2021-06-30:
does the server run with that variable or do you just complain about the wrong linting in vscode?

HaythemLtifi, 2021-06-30:
it does run but i get this error  'Tried to read attribute  #'16' in TwoByteNodeId(i=33), but the attribute is missing'
so i assumed its all connected to the fact that the param type is not node its just any. not sure what is happening here.
thanks for your time man. very much appreciated

closed 2021-06-30 by HaythemLtifi