FreeOpcUa/python-opcua #1352: Client calling opcua methods without arguments throws exception

mrkrizic, 2021-07-07:
**Describe the bug** <br />
When calling a method without return values using call_method() or call_method_full the client throws and exception: 'NoneType' object is not iterable.
When calling the same method using _call_method() no exception occurs. This is due to call_method() calling call_method_full() which in the last step iterates over the OutputArguments. Since there are none, the exception is thrown.

**To Reproduce**<br />
Create a method without return type. In my case I was using a 4diac opcua server which uses the C opcua implementation.

**Expected behavior**<br />
The method call would succeed and None or [] would be returned.

**Screenshots**<br />
This is the method viewed using UA_Expert. It has no input or output parameters (Will be changed in the future).
[image]

**Possible solution**
Check if arguments are not None before collecting the Output Arguments in: https://github.com/FreeOpcUa/python-opcua/blob/3012acf9047aad3acc2674e9008f0d1eee680c25/opcua/common/methods.py#L41

**Version**<br />
Python-Version: 2.7, but should give the same results on any version<br />
python-opcua Version (e.g. master branch, 0.9): 0.98.13

zerox1212, 2021-07-08:
Did you check OPC UA spec if its allowed to have no return arguments? I can't remember.

AndreasHeine, 2021-07-08:
It is allowed to have no args e.g. in Part 10 Programs the statechange method which have no args at all!

mrkrizic, 2021-07-08:
As far as I understood it, you are allowed to have no output arguments. 

According to the [documentation](https://reference.opcfoundation.org/v104/Core/docs/Part4/5.11.2/), methods without output arguments should return an empty list.

AndreasHeine, 2021-07-08:
long things shot could you make a wireshark trace of the method call so we can figure out if the server returns an empty list or not!?

mrkrizic, 2021-07-08:
I will try, but I am currently not in the office so I'll need to come up with something

mrkrizic, 2021-07-19:
I am not able to perform a wireshark trace, but I have found the server function block implementation for the method nodes in my example. It is taken from an amalgated source file of open62514 and is used in 4diac to provide an abstract opcua layer for function blocks. Sorry for the big code block, but I don't know which part might be interesting to you.

```#ifdef UA_ENABLE_METHODCALLS

static const UA_NodeId hasproperty = {0, UA_NODEIDTYPE_NUMERIC, {UA_NS0ID_HASPROPERTY}};
static const UA_NodeId propertytype = {0, UA_NODEIDTYPE_NUMERIC, {UA_NS0ID_PROPERTYTYPE}};

static UA_StatusCode
UA_Server_addMethodNodeEx_finish(UA_Server *server, const UA_NodeId nodeId,
                                 UA_MethodCallback method,
                                 const size_t inputArgumentsSize, const UA_Argument *inputArguments,
                                 const UA_NodeId inputArgumentsRequestedNewNodeId,
                                 UA_NodeId *inputArgumentsOutNewNodeId,
                                 const size_t outputArgumentsSize, const UA_Argument *outputArguments,
                                 const UA_NodeId outputArgumentsRequestedNewNodeId,
                                 UA_NodeId *outputArgumentsOutNewNodeId) {
    /* Browse to see which argument nodes exist */
    UA_BrowseDescription bd;
    UA_BrowseDescription_init(&bd);
    bd.nodeId = nodeId;
    bd.referenceTypeId = UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY);
    bd.includeSubtypes = false;
    bd.browseDirection = UA_BROWSEDIRECTION_FORWARD;
    bd.nodeClassMask = UA_NODECLASS_VARIABLE;
    bd.resultMask = UA_BROWSERESULTMASK_BROWSENAME;

    UA_BrowseResult br;
    UA_BrowseResult_init(&br);
    UA_UInt32 maxrefs = 0;
    Operation_Browse(server, &server->adminSession, &maxrefs, &bd, &br);

    UA_StatusCode retval = br.statusCode;
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Server_deleteNode(server, nodeId, true);
        UA_BrowseResult_deleteMembers(&br);
        return retval;
    }

    /* Filter out the argument nodes */
    UA_NodeId inputArgsId = UA_NODEID_NULL;
    UA_NodeId outputArgsId = UA_NODEID_NULL;
    const UA_QualifiedName inputArgsName = UA_QUALIFIEDNAME(0, "InputArguments");
    const UA_QualifiedName outputArgsName = UA_QUALIFIEDNAME(0, "OutputArguments");
    for(size_t i = 0; i < br.referencesSize; i++) {
        UA_ReferenceDescription *rd = &br.references[i];
        if(rd->browseName.namespaceIndex == 0 &&
           UA_String_equal(&rd->browseName.name, &inputArgsName.name))
            inputArgsId = rd->nodeId.nodeId;
        else if(rd->browseName.namespaceIndex == 0 &&
                UA_String_equal(&rd->browseName.name, &outputArgsName.name))
            outputArgsId = rd->nodeId.nodeId;
    }

    /* Add the Input Arguments VariableNode */
    if(inputArgumentsSize > 0 && UA_NodeId_isNull(&inputArgsId)) {
        UA_VariableAttributes attr = UA_VariableAttributes_default;
        char *name = "InputArguments";
        attr.displayName = UA_LOCALIZEDTEXT("", name);
        attr.dataType = UA_TYPES[UA_TYPES_ARGUMENT].typeId;
        attr.valueRank = UA_VALUERANK_ONE_DIMENSION;
        UA_UInt32 inputArgsSize32 = (UA_UInt32)inputArgumentsSize;
        attr.arrayDimensions = &inputArgsSize32;
        attr.arrayDimensionsSize = 1;
        UA_Variant_setArray(&attr.value, (void*)(uintptr_t) inputArguments,
                            inputArgumentsSize, &UA_TYPES[UA_TYPES_ARGUMENT]);
        retval |= UA_Server_addVariableNode(server, inputArgumentsRequestedNewNodeId, nodeId,
                                            hasproperty, UA_QUALIFIEDNAME(0, name),
                                            propertytype, attr, NULL, &inputArgsId);
    }

[... 95 lines cut ...]
editMethodCallback(UA_Server *server, UA_Session* session,
                   UA_Node* node, void* handle) {
    if(node->nodeClass != UA_NODECLASS_METHOD)
        return UA_STATUSCODE_BADNODECLASSINVALID;
    UA_MethodNode *mnode = (UA_MethodNode*) node;
    mnode->method = (UA_MethodCallback)(uintptr_t)handle;
    return UA_STATUSCODE_GOOD;
}

UA_StatusCode
UA_Server_setMethodNode_callback(UA_Server *server,
                                 const UA_NodeId methodNodeId,
                                 UA_MethodCallback methodCallback) {
    return UA_Server_editNode(server, &server->adminSession, &methodNodeId,
                              (UA_EditNodeCallback)editMethodCallback,
                              (void*)(uintptr_t)methodCallback);
}

#endif ```

There is no output argument refeference node in the first place, since the definition of an outputArgument Node is optional if I understand the reference correctly. Since there is no outputArgument, the check `elif len(result.OutputArguments) == 1:` fails. Therefore nothing was returned, which would leave the value uninitialized to None and the client throws an exception.

UaExpert is able to handle this case and from the documentation it seems to me that this is a valid implementation.

Correct me if I'm wrong.

mrkrizic, 2021-08-03:
This seems related: https://www.gitmemory.com/issue/eclipse/milo/838/839512557