FreeOpcUa/python-opcua #1355: Reading a quite complex extension object results in buffer problems

pridi, 2021-07-08:
### Discussed in https://github.com/FreeOpcUa/python-opcua/discussions/1354

<div type='discussions-op-text'>

<sup>Originally posted by **pridi** July  8, 2021</sup>
Hello everyone, I'm facing an issue when I try to read a structure like this (IcsEvoShifts in the picture):

[image]

When I try to get value before executing the load_type_definitions(), I correctly see in the body the bytes stream:

[image]

but after executing the load_type_definitions() and executing again the get_value(), I receive a Not enough data left in buffer:

_raise NotEnoughData("Not enough data left in buffer, request for {0}, we have {1}".format(size, self))
opcua.common.utils.NotEnoughData: Not enough data left in buffer, request for 296960, we have Buffer(size:700,..._

This is the very simple snipped code I have:

```
client.connect()

struct=client.get_node("s=IcsEvoShifts")
root = client.get_root_node()
objects = client.get_objects_node()
children = root.get_children()
child_of_children=children[0].get_children()
print("child_of_children  :",child_of_children)
c_child_of_children=child_of_children[3].get_children()
print("c_child_of_children  :", c_child_of_children)
structs = client.get_node(c_child_of_children[5])
before = struct.get_value()

client.load_type_definitions()

after = struct.get_value()
```

Could you please explain me what could the issue?

Thanks</div>

oroulet, 2021-07-08:
try with the sync API of asyncua instead of python-opcua . If you still get the issue I need to know the definitoin of that structures.

pridi, 2021-07-08:
Thank you for your quick reply. Could you please help me to translate the previous snipped with asyncua? I'm quite new to it. 

Thanks

oroulet, 2021-07-08:
use asyncua library. The only changes should be to import `from asyncua.sync import XXX` instead of `from opcua` https://github.com/FreeOpcUa/opcua-asyncio

oroulet, 2021-07-08:
aslre replace get_xx with read_xxx

pridi, 2021-07-08:
I tried to use the provided demo code in the async repo. The results is the same.

```
import sys
import time
import logging
import asyncio

from asyncua import Client
from asyncua import ua

async def main():
    logging.basicConfig(level=logging.WARN)
    #async with Client("opc.tcp://asyncua.demo-this.com:51210/UA/SampleServer") as client:
    async with Client("opc.tcp://172.16.4.28:4840/") as client:

        uri = 'urn:unconfigured:application'
        idx = await client.get_namespace_index(uri)

        struct = await client.nodes.objects.get_child(f"{idx}:IcsEvoShifts")
        before = await struct.read_value()
        await client.load_type_definitions()  # scan server for custom structures and import them
        after = await struct.read_value()
        print("before", before)
        print("after", after)

if __name__ == "__main__":
    asyncio.run(main())
```

the error is the same:

[image]

pridi, 2021-07-08:
Ok, now I do not have any more the error but the struct has not been decoded. I still see the bytes stream and not the decoded version.

[image]

Am I missing something?

oroulet, 2021-07-08:
load_data_type_definitoin() is following the 1.04 spec. Your server probably does not support that verison of spec. What server are you using?
Anyway honestly I do not have time debugging older spec version. I do not use them

pridi, 2021-07-09:
For the version I have to ask to my customer (I don't know how to retrieve it with UAExpert...my bad!).
You're right, if I use load_type_definitions() method and I look at the content of the returned variable, I see the complete and correct list of types of my struct. On the other side load_data_type_definitoin() is returning None.
I was expecting that having the types (with load_type_definitions()) the decoding was producing in any case a good results.

[image]

pridi, 2021-07-14:
@oroulet I've changed the PLC with a brand new Codesys PLC (V.3.15). Same structures as with the old PLC but the problem is totally the same.

swamper123, 2021-07-15:
@pridi What OPC UA spec version does it support?

KolbenKeule, 2022-03-31:
Hi, I need help too. I want to connct with an username and password but when I´m trying this with
client.set_user("********")
client.set_password("*********")
I got an error with: 
asyncua.ua.uaerrors._auto.BadIdentityTokenInvalid: "The user identity token is not valid."(BadIdentityTokenInvalid)
Can someone help me please? I don´t know what is wrong.

AndreasHeine, 2022-03-31:
@KolbenKeule why do you post this on a half year old topic about extension objects?

Questions -> https://github.com/FreeOpcUa/python-opcua/discussions

KolbenKeule, 2022-03-31:
Hi @AndreasHeine after your help. I try to code my own script but after 2 weeks I´m at the same positition like befor. Nothing works. :/

AndreasHeine, 2022-03-31:
@KolbenKeule if nothing works its likely a misunderstanding/misuse of the technology.
feel free to open issues or start discussions but remember this is all open source (supported by volunteers) means no one gets payed for that!
i can only encourage you to at least read part 1 of the opc ua spec https://reference.opcfoundation.org/ as a starting point to get the basics and from then start asking the correct questions :wink: (opc ua is nothing you learn in 1h youtube tutorial ^^)