FreeOpcUa/python-opcua #1360: Authentication with NONE security policy

RakshanPremsagar, 2021-07-14:
I am trying to connect to a server, with securitypolicy NONE and having user credentials.  
I get an error at the connect() function.

If we do not mention any security policy, does it mean it takes a default (NONE)?
```
 endpointURL= "opc.tcp://localhost:48020"
 client = Client(endpointURL)
 client.application_uri = "urn:example.org:FreeOpcUa:python-opcua"
 client.set_user("TECH")
 client.set_password("tech")

client.connect() #FAILS
```

Also tried with

```
client.set_security_string( "None,None,my_cert.der,my_private_key.pem")
client.connect() # FAILS 
```

Any guidance is welcome

AndreasHeine, 2021-07-14:
Issuetemplate!
Error? StackTrace?

please stick to the template if you want help!

RakshanPremsagar, 2021-07-14:
Yes sorry, it was a miss from my end. Port number was changed. 
```
 endpointURL= "opc.tcp://localhost:48010"
 client = Client(endpointURL)
 client.application_uri = "urn:example.org:FreeOpcUa:python-opcua"
 client.set_user("TECH")
 client.set_password("tech")

client.connect() # Pass
```

closed 2021-07-14 by RakshanPremsagar