FreeOpcUa/python-opcua #1363: OPC UA SERVEUR only works with  Basic256Sha256_SignAndEncrypt ?

HaythemLtifi, 2021-07-16:
### Discussed in https://github.com/FreeOpcUa/python-opcua/discussions/1362

<div type='discussions-op-text'>

<sup>Originally posted by **HaythemLtifi** July 15, 2021</sup>
Hello,
i wanted to try the Basic256_Sign  SecurityPolicyType byt the UaExpert throw this error
i cant only connect if i set it to Basic256Sha256_SignAndEncrypt  .  i have labview client and i need the Basic256_Sign   to connect with it.
any help plz.
thanks 
[image]
[image]

[image]
</div>

AndreasHeine, 2021-07-16:
duplicate!

closed 2021-07-16 by AndreasHeine

HaythemLtifi, 2021-07-16:
sorry for the duplicate  i thought this is adequate to be and issue more than ask for help section  .