FreeOpcUa/python-opcua #1364: Help with methods with CustomTypes

siscia, 2021-07-16:
Hello,

I am trying to make a very simple server works.

I want a method who takes as input a custom type.

Here below, a reduced example.

The `square` method works perfectly fine.

The one that it is not working is the `sum_of_squares` method.

```python
#!/usr/bin/env python3

import sys

from opcua import ua, Server
from opcua.common.type_dictionary_buider import DataTypeDictionaryBuilder

def square(parent, variant):
    print("square method call with parameters: ", variant.Value)
    variant.Value *= variant.Value
    return [variant]

def sum_of_squares(parent, variant):
    print("sum_of_square methos call with parameter: ", variant.Value)
    ## res = variant.Value.i**2 + variant.Value.j**2 # <- I was expecting this to work, somehow
    ret = 3*3 + 8*8
    return [ua.Variant(ret, ua.VariantType.Int64)]

if __name__ == "__main__":
    server = Server()
    server.create_custom_data_type(2, "ComplexType", description="A complex type")

    server.set_endpoint("opc.tcp://0.0.0.0:4840/")

    ns = server.register_namespace("http://gopcua.com/")
    dictBuilder = DataTypeDictionaryBuilder(server, ns, "http://gopcua.com/", "TestTypeBuilder")
    typeName = "ComplexType"
    dictBuilder.create_data_type(typeName)
    dictBuilder.add_field("i", ua.VariantType.Int64, typeName)
    dictBuilder.add_field("j", ua.VariantType.Int64, typeName)
    dictBuilder.set_dict_byte_string()
    server.load_type_definitions()
    
    main = server.nodes.objects.add_object(ua.NodeId("main", ns), "main")
    fnSquare = main.add_method(ua.NodeId("square", ns), "square", square, [ua.VariantType.Int64], [ua.VariantType.Int64])
    fnSumOfSquare = main.add_method(ua.NodeId("sumOfSquare", ns), "sumOfSquare", sum_of_squares, [ua.VariantType.ExtensionObject], [ua.VariantType.Int64])

    server.start()
```

Out of which I got this log.

```
sum_of_square methos call with parameter:  ExtensionObject(TypeId:StringNodeId(ns=2;s=ComplexType), Encoding:1, 16 bytes)
```

So the object is sent correctly from the client.

Any idea on what I am doing wrong?

AndreasHeine, 2021-07-16:
not an issue! please use the Discussions feature: https://github.com/FreeOpcUa/python-opcua/discussions

closed 2021-07-16 by AndreasHeine

siscia, 2021-07-16:
Aw, I was not familiar with this new feature of github!

siscia, 2021-07-16:
Thanks, opened https://github.com/FreeOpcUa/python-opcua/discussions/1366