FreeOpcUa/python-opcua #1365: Set AccessLevel for properties

deretive, 2021-07-16:
Hey guys, I am pretty new to OPC UA and try to set the AccessLevel for Nodes from the Machinery XML. 
I am able to set the WriteMask for the Node itself, but I cannot set it for the subordinated properties.
` emergencyStopFunction.active.set_attribute(ua.AttributeIds.AccessLevel, ua.DataValue(ua.Variant(2, ua.VariantType.UInt32)))`
It keeps telling
AttributeError: 'bool' object has no attribute 'set_attribute'

Probably this is easy, but I really struggle to find the solution.

AndreasHeine, 2021-07-16:
do you mean setting a propertie writeable?
https://github.com/FreeOpcUa/python-opcua/blob/3012acf9047aad3acc2674e9008f0d1eee680c25/opcua/common/node.py#L221

AndreasHeine, 2021-07-16:
or is this a typo: `emergencyStopFunction.active.set_attribute` -> `emergencyStopFunction.set_attribute`

deretive, 2021-07-16:
yes, but `emergencyStopFunction.active.set_writable()` does not work too ("AttributeError: 'bool' object has no attribute 'set_writable' "), I already tried it this way, because other (not imported) variables are working this way. 

`emergencyStopFunction.set_attribute` is denied with the error "BadAttributeIdInvalid: "The attribute is not supported for the specified Node."(BadAttributeIdInvalid)"

AndreasHeine, 2021-07-16:
"AttributeError: 'bool' object has no attribute 'set_writable' " means that `emergencyStopFunction.active` is not a node-calss instance so it for sure has no method `set_writable()` ...

you propably debug that with breakpoints ans inspect the objects!

deretive, 2021-07-16:
Spyder Object Inspector says for everything that there is no further documentation available.
Maybe this code snipped helps a bit:
`class EmergencyStopFunction:    emergencyStopFunction_nodeid=Server.server.get_node(ua.ObjectIds.BaseObjectType).get_child(["3:EmergencyStopFunctionType"])
    emergencyStopFunction=MyFolderFolder.add_object('ns=1;s="EmergencyStopFunction"', "EmergencyStopFunction", emergencyStopFunction_nodeid)
    emergencyStopFunction.set_attribute(ua.AttributeIds.WriteMask, ua.DataValue(ua.Variant(2, ua.VariantType.UInt32)))
    emergencyStopFunction.active= False  
    emergencyStopFunction.active.set_attribute(ua.AttributeIds.AccessLevel, ua.DataValue(ua.Variant(2, ua.VariantType.UInt32)))`

deretive, 2021-07-16:
Spyder Object Inspector says for everything that there is no further documentation available.
Maybe this code snipped helps a bit:
`class EmergencyStopFunction:
    emergencyStopFunction_nodeid=Server.server.get_node(ua.ObjectIds.BaseObjectType).get_child(["3:EmergencyStopFunctionType"])
    emergencyStopFunction=MyFolder.add_object('ns=1;s="EmergencyStopFunction"', "EmergencyStopFunction", emergencyStopFunction_nodeid)
    emergencyStopFunction.set_attribute(ua.AttributeIds.WriteMask, ua.DataValue(ua.Variant(2, ua.VariantType.UInt32)))
    emergencyStopFunction.active= False  
    emergencyStopFunction.active.set_attribute(ua.AttributeIds.AccessLevel, ua.DataValue(ua.Variant(2, ua.VariantType.UInt32)))`

AndreasHeine, 2021-07-16:
`emergencyStopFunction.active= False`

`emergencyStopFunction.active.set_attribute(ua.AttributeIds.AccessLevel, ua.DataValue(ua.Variant(2, ua.VariantType.UInt32)))`
"AttributeError: 'bool' object has no attribute 'set_writable' "

the interpreter/compiler is always right ^^

set_writable and set_attribute are node class methods and can only be called on node class instances...
this appearce to be a node class instance: `emergencyStopFunction_nodeid` or `emergencyStopFunction`

node class definition in src:
https://github.com/FreeOpcUa/python-opcua/blob/3012acf9047aad3acc2674e9008f0d1eee680c25/opcua/common/node.py#L31

deretive, 2021-07-16:
Thanks a lot, your hints helped me to figure out how it works :)
In case someone else struggles with this too, here is my solution:
`class EmergencyStopFunction:
    emergencyStopFunction_nodeid=Server.server.get_node(ua.ObjectIds.BaseObjectType).get_child(["3:EmergencyStopFunctionType"])
    emergencyStopFunction=MyFolder.add_object('ns=1;s="EmergencyStopFunction"', "EmergencyStopFunction", emergencyStopFunction_nodeid)
    emergencyStopFunctionActive=Server.server.get_root_node().get_child(["0:Objects","4:Machines","0:MyFolder","0:EmergencyStopFunction","3:Active"])
    emergencyStopFunctionActive.set_attribute(ua.AttributeIds(13), ua.DataValue(False))
    emergencyStopFunctionActive.set_writable()
    emergencyStopFunctionActive=Server.server.get_root_node().get_child(["0:Objects","4:Machines","0:MyFolder","0:EmergencyStopFunction","3:Name"])
    emergencyStopFunctionActive.set_attribute(ua.AttributeIds(13), ua.DataValue("KillSwitch"))
    emergencyStopFunctionActive.set_writable()`

closed 2021-07-16 by deretive