FreeOpcUa/python-opcua #1367: Issues creating subscription

apoorvcn47, 2021-07-22:
Hi,
I am trying to run this example:
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py

This example crashes with following log

1626973824.609421 [0]    python3: using network interface enp40s0 (udp/192.168.0.200) selected arbitrarily from: enp40s0, wlp0s20f3
Root node is:  i=84
Objects node is:  i=85
Children of root are:  [Node(TwoByteNodeId(i=87)), Node(TwoByteNodeId(i=86)), Node(TwoByteNodeId(i=85))]
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response
Traceback (most recent call last):
  File "/home/aw_rdex/rr_ws/src/pallet_optimization/po_fanuc_driver/client.py", line 92, in <module>
    sub.subscribe_events()
  File "/home/aw_rdex/.local/lib/python3.8/site-packages/opcua/common/subscription.py", line 216, in subscribe_events
    return self._subscribe(sourcenode, ua.AttributeIds.EventNotifier, evfilter, queuesize=queuesize)
  File "/home/aw_rdex/.local/lib/python3.8/site-packages/opcua/common/subscription.py", line 234, in _subscribe
    mids[0].check()
  File "/home/aw_rdex/.local/lib/python3.8/site-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadAttributeIdInvalid: "The attribute is not supported for the specified Node."(BadAttributeIdInvalid)

When i try to comment out line 84 (sub.subscribe_events()), the program seems to work. but i only get value_change_event notification once per run. nothing after that.

Any ideas?

swamper123, 2021-07-23:
### Please read, before you post!

This is a _**BUG REPORT for issues in the existing code**_. 

If you have general questions, code handling problems or ideas, please use the:

Discussionsboard: https://github.com/FreeOpcUa/python-opcua/discussions <br/> or <br />
Gitter-Channel: https://gitter.im/FreeOpcUa/python-opcua

---

To save some time, please provide us following informations, if possible:

**Describe the bug** <br />
A clear and concise description of what the bug is.

**To Reproduce**<br />
Steps to reproduce the behavior incl code. <br />
If you post code, please use the "insert Code" (which is the "< >" symbol) format for a better readability.

**Expected behavior**<br />
A clear and concise description of what you expected to happen.

**Screenshots**<br />
If applicable, add screenshots to help explain your problem. <br />
You can post Screenshots directly to github, please don't use 3rd Party webpages

**Version**<br />
Python-Version:<br />
python-opcua Version (e.g. master branch, 0.9):

swamper123, 2021-07-23:
What I want to say is, that you should describe your setup: What Server are you using? What Node are you trying to subscribe? What version of python-opcua are you using? etc

oroulet, 2021-07-23:
Also I that particular case your server is not supporting the EventNotifier attribute as the error clearly states.