FreeOpcUa/python-opcua #1368: issue importing UA/Pumps from official UA-Nodesets

marvin-meck, 2021-07-27:
Hi there, 

I'm fairly new to OPC UA in general, so I'm unsure if I'm missing anything obvious. Just in case, apologies in advance! 

I was trying to import the official [UA/Pumps](https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04/Pumps) node set using the ```import_xml``` method the other day. 
Importing the required models (DI and Machinery) works just fine. However, with the Pumps node set I keep getting the following exception that I don't know how to deal with: 

> Exception: ('Error no extension class registered ', 'PhysicalAddressDataType', NumericNodeId(ns=4;i=3020)) 

From what I understand, the companion specification for pumps defines a new data type for physical addresses, but the type is somehow unknown or isn't added at any point. I'm also uncertain whether the issue lies with the ```python-opcua``` package or the official node sets. 

Help would be much appreciated!

Example Code:
```
import opcua

address="opc.tcp://127.0.0.1:12345"
    
server = opcua.Server()
server.set_endpoint(address)

node_set_list = [
    "./UA-Nodeset/DI/Opc.Ua.Di.NodeSet2.xml", 
    "./UA-Nodeset/Machinery/Opc.Ua.Machinery.NodeSet2.xml", 
    "./UA-Nodeset/Pumps/Opc.Ua.Pumps.NodeSet2.xml"
]
    
for node_set in node_set_list:
    server.import_xml(node_set)
    print(f"successfully imported {node_set}")
```

AndreasHeine, 2021-08-19:
can you post the debug output?

`logger.setLevel(logging.DEBUG)`

marvin-meck, 2021-09-02:
Hey, sorry for the delay! I just saw that reply the other day! 
[pumps.log](https://github.com/FreeOpcUa/python-opcua/files/7097221/pumps.log)

janfelixklein, 2021-11-03:
I currently have exactly the same problem when importing the [I4AAS Nodeset](https://github.com/OPCFoundation/UA-Nodeset/blob/v1.04/I4AAS/Opc.Ua.I4AAS.NodeSet2.xml) which also has a customn datatype which is refered to inside the spec itself and is throwing the same Exception

AndreasHeine, 2021-11-03:
you propably need to switch to opcua-asyncio! -> https://github.com/FreeOpcUa/opcua-asyncio

janfelixklein, 2021-11-04:
The problem also occurs when using opcua-asyncio. 

I guess what happens is that at one point the nodeset defines a customn structure as a `DataType` definition, e.g. node `ns=1;i=3011 1:AASKeyDataType` in the I4AAS nodeset. Later, this datatype is used in a variable, e.g. `ns=1;i=6003 1:Keys`.

However, the xmlimporter is does not register an extension object when a new structure is created. I guess as soon as a customn structure is defined there needs to be some routine like the (`server.load_type_definitions() `) to create an extension object.

oroulet, 2021-11-04:
ok. looks like the logic has to be changed here. It is not 100% straightforward. any PR welcome. I am not using the xml importa currently

anshi43, 2023-06-08:
Hello @janfelixklein , @AndreasHeine ,@marvin-meck and @oroulet ,
did you guys find the solution to this issue? because I desperately need the solution as soon as possible. 

when i am import the pump nodeset file it throws the same error : **Exception: ('Error no extension class registered ', 'PhysicalAddressDataType', NumericNodeId(ns=4;i=3020))**

whom should i contact in order to solve this error? 

your advice and solution would be appreciated.

oroulet, 2023-06-08:
First use opcua-asyncio

anshi43, 2023-06-08:
> First use opcua-asyncio

i tried but it throws an error : 

here is the code :

`from asyncua import ua, Server
import asyncio

async def main():
    server = Server()
    await server.init()     
    server.set_endpoint("opc.tcp://10.221.0.107:4840")
    #await server.import_xml("Opc.Ua.Nodeset2.xml")
    await server.import_xml("Opc.Ua.Di.Nodeset2.xml")
    await server.import_xml("Opc.Ua.Machinery.Nodeset2.xml")
    await server.import_xml("Opc.Ua.Pumps.Nodeset2.xml")
    

    await server.start()

if __name__ == "__main__":
    asyncio.run(main())`

here is the error :

NodeData(nodeid:NodeId(Identifier=333, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>)) has datatypedefinition and path [Node(NumericNodeId(Identifier=84, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NumericNodeId(Identifier=86, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NodeId(Identifier=90, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>)), Node(NumericNodeId(Identifier=24, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NumericNodeId(Identifier=26, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NumericNodeId(Identifier=28, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NodeId(Identifier=7, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>))] but we could not find out if this is a struct
renamed AS-Interface to AS_Interface due to Python syntax
renamed BACnet/IP to BACnet_IP due to Python syntax
renamed BACnet/MSTP to BACnet_MSTP due to Python syntax
renamed CC-Link to CC_Link due to Python syntax
renamed Ethernet/IP to Ethernet_IP due to Python syntax
renamed EthernetTCP/IP to EthernetTCP_IP due to Python syntax
renamed IO-Link to IO_Link due to Python syntax
renamed LIN-Bus to LIN_Bus due to Python syntax
renamed LTE-M to LTE_M due to Python syntax
renamed M-Bus to M_Bus due to Python syntax
renamed MP-Bus to MP_Bus due to Python syntax
renamed NB-IOT to NB_IOT due to Python syntax
renamed X2X-Link to X2X_Link due to Python syntax
renamed Z-Wave to Z_Wave due to Python syntax
renamed 2006_42_EC to _2006_42_EC due to Python syntax
renamed 2009_125_EC to _2009_125_EC due to Python syntax
renamed 2011_65_EU to _2011_65_EU due to Python syntax
renamed 2014_35_EU to _2014_35_EU due to Python syntax
renamed 2014_34_EU to _2014_34_EU due to Python syntax
renamed 2014_30_EU to _2014_30_EU due to Python syntax
renamed 2014_68_EU to _2014_68_EU due to Python syntax
renamed 2014_29_EU to _2014_29_EU due to Python syntax
renamed 1G to _1G due to Python syntax
renamed 2G to _2G due to Python syntax
renamed 3G to _3G due to Python syntax
renamed 1D to _1D due to Python syntax
renamed 2D to _2D due to Python syntax
renamed 3D to _3D due to Python syntax
renamed Zone 0 to Zone_0 due to Python syntax
renamed Zone 1 to Zone_1 due to Python syntax
renamed Zone 2 to Zone_2 due to Python syntax
renamed Zone 20 to Zone_20 due to Python syntax
renamed Zone 21 to Zone_21 due to Python syntax
renamed Zone 22 to Zone_22 due to Python syntax
renamed Constant pressure control to Constant_pressure_control due to Python syntax
renamed Constant temperature control to Constant_temperature_control due to Python syntax
renamed Differential pressure control to Differential_pressure_control due to Python syntax
renamed Constant differential pressure control to Constant_differential_pressure_control due to Python syntax 
renamed Variable differential pressure control to Variable_differential_pressure_control due to Python syntax
renamed Flow_dependent differential pressure control to Flow_dependent_differential_pressure_control due to Python syntax
renamed Return flow temperature control to Return_flow_temperature_control due to Python syntax
renamed Flow temperature control to Flow_temperature_control due to Python syntax
renamed Flow rate control to Flow_rate_control due to Python syntax
renamed Speed control to Speed_control due to Python syntax
renamed Bluetooth Low Energy to Bluetooth_Low_Energy due to Python syntax
renamed DECT ULE to DECT_ULE due to Python syntax
renamed Ethernet TCP_IP to Ethernet_TCP_IP due to Python syntax
renamed Modbus TCP to Modbus_TCP due to Python syntax
renamed Modbus RTU to Modbus_RTU due to Python syntax
renamed OPC UA to OPC_UA due to Python syntax
renamed OPC DA to OPC_DA due to Python syntax
renamed PROFIBUS DP to PROFIBUS_DP due to Python syntax
renamed PROFINET RT to PROFINET_RT due to Python syntax
Endpoints other than open requested but private key and certificate are not set.