FreeOpcUa/python-opcua #1370: Regarding getting WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response & opcua.ua.uaerrors._auto.BadSessionIdInvalid: "The session id is not valid."(BadSessionIdInvalid)

rajnishnationfirst, 2021-08-12:
Hi ,

I am facing too much issue of the type after some time in the client-minimal example .

```
if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    # logger = logging.getLogger("KeepAlive")
    # logger.setLevel(logging.DEBUG)

    client = Client("opc.tcp://XXXXXXXX:4861")

    try:
        client.connect()

        # Client has a few methods to get proxy to UA nodes that should always be in address space such as Root or Objects
        root = client.get_root_node()
        print("Root node is: ", root)
        objects = client.get_objects_node()
        print("Objects node is: ", objects)

        node_Id_List = []
        for iterator in range(len(tags_node_ids_list)):
            node_Id_List.append(client.get_node(tags_node_ids_list[iterator]))

  
        # Now getting a variable node using its browse path

        myvar = client.get_node("ns=1;s=t|HMI_Tag_1")

        # subscribing to a variable node
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_data_change(node_Id_List)
        time.sleep(0.1)

```
I am subscribing full list of 1000 tags / nodes and i get the values and after sometimes automatatcially i recived the below errors and exception.

How can i overcome the below exceptions .. pls help .

**alarm_was_rasied = %s WHERE work_order_id = %s ('EA08BD_WO_6956', '{"AF400": {"ST_AF400_Maintenance_Call": 0, "ST_AF400_Safety_Nok(Emg.Stop)": 0, "ST_AF400_Station_Sequence_Fault": 1, "ST_AF400_Air_Pressure_Nok": 0, "ST_AF400_Station_Power_Failure": 0, "ST_AF400_Station_Communication_Fault": 0, "ST_AF400_Conveyor_Fault": 0}}', 6956)
NEW WORK CREATION IF BREAKDOWN EXISTS BUT CLOSED
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response
ERROR:concurrent.futures:exception calling callback for <Future at 0x268319208e0 state=finished returned Buffer>
Traceback (most recent call last):
  File "concurrent\futures\_base.py", line 329, in _invoke_callbacks
  File "opcua\client\ua_client.py", line 493, in _call_publish_callback
  File "opcua\client\ua_client.py", line 93, in check_answer
  File "opcua\ua\uatypes.py", line 218, in check
opcua.ua.uaerrors._auto.BadSessionIdInvalid: "The session id is not valid."(BadSessionIdInvalid)
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response
ERROR:concurrent.futures:exception calling callback for <Future at 0x268319206a0 state=finished returned Buffer>
Traceback (most recent call last):
  File "concurrent\futures\_base.py", line 329, in _invoke_callbacks
  File "opcua\client\ua_client.py", line 493, in _call_publish_callback
  File "opcua\client\ua_client.py", line 93, in check_answer
  File "opcua\ua\uatypes.py", line 218, in check
opcua.ua.uaerrors._auto.BadSessionIdInvalid: "The session id is not valid."(BadSessionIdInvalid)**

AndreasHeine, 2021-08-12:
looks like the session dies... what happens on the devices ore the network (reboots, timeouts, applikation restart, ...)

if you would have filled out the issue template it would be easier to help you!

swamper123, 2021-08-12:
What does "some time" mean? An hour? 10 minutes?
How do you handle connection errors?
What Hardware & Software are you using? ....

rajnishnationfirst, 2021-08-12:
Thanks @oroulet  AndreasHeine,

I take the nearly 1000 and suscribe with a HANDLER class for ondatachange call using SubHandler .

I get the ondatachange executed for some tags on their value change.

What happned a few days back , few tags were added and connected to PLC and then after this issue arised and still exists .

Earlier with 936 nodes it was working.

Pls help me how to overcome this problem . If you can elaborate on the template part  , what you are rfering above if my replies does'nt answer your questions. 

Pls help AndreasHeine , i m really worried .

rajnishnationfirst, 2021-08-12:
swamper123 thanks for your reply .

We have SIEMENS PLC attached with the WINCC  Server  and i am subscribing using feeopcua python client. But i am not able to resolve it . Pls help me anyone i m worried.

swamper123, 2021-08-12:
@rajnishnationfirst
1) you should fill out the template properly, so we could help you even better, so we know the setup better you have.
2) What are the max. Serverlimits? What does the instruction-manual or server say?
3) Again the question:
What does "some time" mean? An hour? 10 minutes? What have you observed?
How do you handle connection errors in your client code?

closed 2021-08-19 by AndreasHeine