FreeOpcUa/python-opcua #1375: call_method function always returns variables of type string

safarsaitam, 2021-09-08:
I have a working OPC-UA server up and running with an available method that multiplies an integer with a float, returning a float. The types of the arguments are set up correctly, both for the inputs and outputs, but whenever I call the `call_method` function, the return value is a string.

Excerpt of method calling function:
```
client = Client(server_url)
            
try:
    return_value = [None, event_value, '']
    client.connect()
    client.load_type_definitions()
    
    root = client.get_root_node()

    uri = "http://systec.fe.up.pu"
    idx = client.get_namespace_index(uri)
    
    folder = root.get_child(["0:Objects", "{}:DINASORE OPC-UA".format(idx), "{}:OPC-UA_Methods".format(idx)])
    res = folder.call_method("{}:{}".format(idx, method_name), *input_list)  
    
    return_value = [None, event_value, res]
    print('Res: {} type {}'.format(res, type(res)))        # Res: 10.0 type <class 'str'>

finally:
    client.disconnect()
    return return_value
```

The method is generated dynamically so I believe it is too complex for me to paste it here. I have a definition file for the method where I indicate the type of its arguments and my application reads the file and generates the method. Here is a screenshot I took when using Prosys to demonstrate the correct type assignment:

[image]
(The value for the 'RUN' property is merely a 1 to fire the multiplication event)

To sum up, the excerpt of code should be printing that the type of the return variable is a float. I analyzed the code from the FreeOpcUa framework but could not find what might be causing this. I also can't simply cast to float because I will be creating and using different methods with different return types.  

Python-Version: 3.9.7
python-opcua Version: master branch

AndreasHeine, 2021-09-08:
what is res at all if you dump it to the console?

safarsaitam, 2021-09-08:
You mean if I write `print(res)`? A straight up `10.0` is displayed.

AndreasHeine, 2021-09-08:
ok and if you add a breakpoint after the call what is res look like in the debugger? still a string variable?

AndreasHeine, 2021-09-08:
in your code sniped the inargs are missing -> on purpose?

hint (ignore the async syntax api should be identical):
https://github.com/AndreasHeine/opcua-tutorial/blob/9fe44669622129ec3c62daed15b70618daa9b88a/client/call_method.py#L52

safarsaitam, 2021-09-08:
It is on purpose. `input_list` is a list with my arguments, my "inargs" so to say. I believe I have to add the asterisk (*) before the list when passing it as argument so as to make everything work.

AndreasHeine, 2021-09-08:
*[] unpacks the list into items i guess ^^ so yes

inarg1 = ua.Variant(200, ua.VariantType.Int64)
inarg2 = ua.Variant(13.0, ua.VariantType.Float)
inarg3 = ua.Variant("1", ua.VariantType.String)
result = method_parent.call_method(method_node, inarg1, inarg2, inarg3)

you could try to try this above to make sure everything is setup properly (args as Variants)!

safarsaitam, 2021-09-08:
Yes, the result is the same if I do it the way you just recommended

AndreasHeine, 2021-09-08:
and in debuger its a string???

safarsaitam, 2021-09-08:
Still trying to use the debugger because sadly I never use it...

AndreasHeine, 2021-09-08:
[image]

safarsaitam, 2021-09-08:
I am very sorry. Upon closer inspection I realize that for whatever reason that I have now forgotten, I was casting each output to a string when adding it to the method. Thank you for your time and help!

closed 2021-09-08 by safarsaitam