FreeOpcUa/python-opcua #1387: load_type_definitions don't load custom types

cbudai, 2021-09-25:
I need to write a data structure in the PLC, but i have a error message, it does not find the DesgloseSKU attribute, so I think that it is not loading the custom types from to the PLC:

**The code that I use**<br />

`from opcua import Client, ua`

`client = Client("opc.tcp://ipAddressOPCserver")`
`client.connect()`
`root = client.get_root_node()`
`objects = client.get_objects_node()`
`client.load_type_definitions()`

`sku = ua.uaprotocol_auto.DesgloseSKU()`
`sku.BarCode = ua.LocalizedText("XX123456789", "en-US")`
`sku.lot = ua.LocalizedText("C24SE232021", "en-US")`
...
...
`dv = ua.DataValue(ua.Variant(sku, ua.VariantType.ExtensionObject))`
`client.get_node('ns=4;s=SKUAssign').set_value(dv)`

`client.disconnect()`

**Version:**<br />
python              3. 9. 7
opcua               0.98.13
opcua-client        0.8.0

robbudge, 2021-10-30:
I am also working on this, and having the a similar issue but with a little more information.
PLC is a Wago 750 Series.
A test program with locally declared datatypes does work correctly with the datatypes loaded and reported correctly.
The problem.
if I utilize a data type from a library added to the project then the load_type_definitions fail.

root =  i=84
Children of root are:  [Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86)), Node(TwoByteNodeId(i=87))]
objects = i=85
Load Structures
Traceback (most recent call last):
  File "/Users/robertbudgen/PycharmProjects/pythonProject2/main.py", line 65, in <module>
    client.load_type_definitions()  # scan server for custom structures and import them
  File "/Users/robertbudgen/PycharmProjects/Cayenne/venv/lib/python3.8/site-packages/opcua/client/client.py", line 626, in load_type_definitions
    return load_type_definitions(self, nodes)
  File "/Users/robertbudgen/PycharmProjects/Cayenne/venv/lib/python3.8/site-packages/opcua/common/structures.py", line 263, in load_type_definitions
    generator.make_model_from_string(xml)
  File "/Users/robertbudgen/PycharmProjects/Cayenne/venv/lib/python3.8/site-packages/opcua/common/structures.py", line 146, in make_model_from_string
    self._make_model(obj)
  File "/Users/robertbudgen/PycharmProjects/Cayenne/venv/lib/python3.8/site-packages/opcua/common/structures.py", line 178, in _make_model
    field.value = get_default_value(field.uatype, enums)
  File "/Users/robertbudgen/PycharmProjects/Cayenne/venv/lib/python3.8/site-packages/opcua/common/structures.py", line 42, in get_default_value
    val = list(getattr(ua, uatype).__members__)[0]
IndexError: list index out of range

The datatypes are declared in basically the same manner with one slight difference
Simple local structures are declared in the following format 
|tprop|WAGO.Application.Simple_DUT
Library structures declared as 
|tprop|WAGO.Application.Burkert.Contactor_DUT

if anyone wants access to the OPCua server to test, it is setup on its own OpenVpn network, access can be granted

AndreasHeine, 2021-10-30:
`The library is in maintenance mode. The plan is now to focus on the asyncio version of this library: opcua-asyncio which also has a sync-wrapper, with very few changes in API`

asyncua has a load_data_type_definitions method which should work!

closed 2021-10-30 by AndreasHeine