FreeOpcUa/python-opcua #1403: Connect via Username and Password while setting Policy and Mode is required

alexanderpetrenz, 2021-10-13:
Hi,

I'm trying to connect to a server where I need to provide username and password. I also need to provide a security policy and a message security mode (see below).

[image]

If I don't provide policy and security mode I'm getting an error saying "Error 'BadServiceUnsupported' was returned during CreateSession".

When using the Library I can provide policy and mode by using the methods `set_security_string` or `set_security`. But this doesn't work: Both methods require to specify a path to certificate + key. If I leave the corresponding path parameters empty the library tries to load the certificate from path `.`.

How am I supposed to use the library in my case?

AndreasHeine, 2021-10-13:
in order to use encryption you will need something to encrypt with and if you want to sign messages you need something to sign with! 
key.pem and cert.pem is required...

AndreasHeine, 2021-10-13:
https://github.com/FreeOpcUa/python-opcua/issues/1033

closed 2021-10-13 by alexanderpetrenz