FreeOpcUa/python-opcua #1422: Issue with connecting via certificate

MalteReiSr, 2021-12-02:
Hi,
I'm trying to connect to our test server and I need a self-signed certificate for that. It works fine in UAExpert.
Theses are the settings there.

[image]

And this is the code I mostly took from the examples.
``` 
from asyncua.sync import Client  # type: ignore
from asyncua.sync import ua  # type: ignore

from typing import Any, Union, Dict, Tuple
import datetime
import time

from cryptography import x509
from cryptography.hazmat.primitives import serialization
from cryptography.hazmat.primitives.asymmetric import rsa
from cryptography.x509.oid import NameOID
from cryptography.hazmat.primitives import hashes

aIP = r"opc.tcp://c200110:48400/UA/ComServerWrapper1"

def createCerts():
    # Generate our key
    key = rsa.generate_private_key(
        public_exponent=65537,
        key_size=2048,
    )
    # Write our key to disk for safe keeping
    with open(".\key.pem", "wb") as f:
        f.write(key.private_bytes(
            encoding=serialization.Encoding.PEM,
            format=serialization.PrivateFormat.TraditionalOpenSSL,
            encryption_algorithm=serialization.NoEncryption(),
        ))
    
    # Generate Cert
    subject = issuer = x509.Name([
        x509.NameAttribute(NameOID.COUNTRY_NAME, u"de"),
        x509.NameAttribute(NameOID.STATE_OR_PROVINCE_NAME, u"Ni"),
        x509.NameAttribute(NameOID.LOCALITY_NAME, u"Goe"),
        x509.NameAttribute(NameOID.ORGANIZATION_NAME, u"companyS"),
        x509.NameAttribute(NameOID.COMMON_NAME, u"Reiner"),
    ])
    cert = x509.CertificateBuilder().subject_name(
        subject
    ).issuer_name(
        issuer
    ).public_key(
        key.public_key()
    ).serial_number(
        x509.random_serial_number()
    ).not_valid_before(
        datetime.datetime.utcnow()
    ).not_valid_after(
        # Our certificate will be valid for 30 days
        datetime.datetime.utcnow() + datetime.timedelta(days=30)
    ).add_extension(
        x509.SubjectAlternativeName([x509.DNSName(u"localhost")]),
        critical=False,
    # Sign our certificate with our private key
    ).sign(key, hashes.SHA256())
    # Write our certificate out to disk.
    with open(".\cert.pem", "wb") as f:
        f.write(cert.public_bytes(serialization.Encoding.PEM))

    print(cert)
    print(key)

if __name__ == "__main__":
    createCerts()
    with Client(aIP) as client:
        client.application_uri="urn:c200110:UA:ComServerWrapper"
        client.set_security_string("Basic256Sha256,Sign,./cert.pem,./key.pem")
        client.connect()
        time.sleep(5)
        print("connected")
        client.disconnect()
```
 
and this is the error I get:
```   
ServiceFault (BadSecurityPolicyRejected, diagnostics: DiagnosticInfo(SymbolicId=None, NamespaceURI=None, Locale=0, LocalizedText=None, AdditionalInfo=None, InnerStatusCode=None, InnerDiagnosticInfo=None)) from server received  in response to CreateSessionRequest
Traceback (most recent call last):
File "test.py", line 79, in <module>
with Client(aIP) as client:
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\sync.py", line 254, in __enter__
self.connect()
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\sync.py", line 94, in wrapper
result = self.tloop.post(aio_func(*args, **kwargs))
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\sync.py", line 52, in post
return futur.result()
File "C:\Users\user\Anaconda3\lib\concurrent\futures\_base.py", line 439, in result
return self.__get_result()
File "C:\Users\user\Anaconda3\lib\concurrent\futures\_base.py", line 388, in __get_result
raise self._exception
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\client\client.py", line 247, in connect
await self.create_session()
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\client\client.py", line 370, in create_session
response = await self.uaclient.create_session(params)
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\client\ua_client.py", line 297, in create_session
data = await self.protocol.send_request(request)
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\client\ua_client.py", line 152, in send_request
self.check_answer(data, f" in response to {request.__class__.__name__}")
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\client\ua_client.py", line 161, in check_answer
hdr.ServiceResult.check()
File "C:\Users\user\Anaconda3\lib\site-packages\asyncua\ua\uatypes.py", line 318, in check
raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadSecurityPolicyRejected: "The security policy does not meet the requirements set by the server."(BadSecurityPolicyRejected)
```

I think I'm missing something. Do I need to add NamespaceURI etc. ? But even then I haven't found anything in the docs or any similar questions. 
Thanks for reading

AndreasHeine, 2021-12-02:
your test server rejects your cert! we as "client" just raise the error from the server response.
have you tried using openssl to create the cert?

MalteReiSr, 2021-12-03:
Thanks a lot @AndreasHeine , I tried it with openssl via the git bash and got the same error again:
```
ServiceFault (BadSecurityPolicyRejected, diagnostics: DiagnosticInfo(SymbolicId=None, NamespaceURI=None, Locale=0, LocalizedText=None, AdditionalInfo=None, InnerStatusCode=None, InnerDiagnosticInfo=None)) from server received  in response to CreateSessionRequest 
```
Even though I'm not 100% sure if I used the correct parameters now, I did it with:
```
openssl req -newkey rsa:2048 -nodes -keyout key.pem -x509 -days 365 -out cert.pem -sha256
```

AndreasHeine, 2021-12-08:
> Edit: we just "fixed" this by changing the security settings on the testserver but still don't get why the certificate isn't working.

what server is it python? or what product?

floriandorre, 2021-12-14:
From what i understood by looking at the code i had the same issue where by using `set_security_string` i was not able to do a certificate connection.

It seems that `set_security_string `does not set in any way the attribute `user_certificate` from the `Client` class. This attribute is used when calling `self.activate_session(username=self._username, password=self._password, certificate=self.user_certificate)` in the `connect` methods and at this point by using only `set_security_string` this attribute will not be set.

The fix for me was to also call `load_client_certificate` and `load_private_key`

Not sure if this will fix your problem @MalteReiSr but it was one of mine when I first tried to do a certificate connection attempt.

AndreasHeine, 2021-12-14:
> It seems that set_security_string does not set in any way the attribute user_certificate from the Client class.

it sets the cert and key for the MessageSecurity, not the Auth part!

```python
def set_security_string(self, string):
        """
        Set SecureConnection mode. String format:
        Policy,Mode,certificate,private_key[,server_private_key]
        where Policy is Basic128Rsa15, Basic256 or Basic256Sha256,
            Mode is Sign or SignAndEncrypt
            certificate, private_key and server_private_key are
                paths to .pem or .der files
        Call this before connect()
        """
```

floriandorre, 2021-12-14:
Does this mean that the cert used for MessageSecurity needs to be different than the one used for auth part ?

floriandorre, 2021-12-14:
> it sets the cert and key for the MessageSecurity, not the Auth part!

Thank you @AndreasHeine this message just solved days of struggling with this !