FreeOpcUa/python-opcua #1425: Regarding Windows Server is getting hanged from last 1 week.

rajnishnationfirst, 2021-12-13:
Hi  colleagues,

There are 2 servers 149  ( WinCC SCADA Server ) and 151 (  Web Server ).

The 151 server i having a OPCUA Client written in Python Free OPCUA . 

This client was running fine from last 2 years , but from last week the Web Server ( 151 ) is getting hanged when i run the same program and i am not able launch it via RDP ( Remote Desktop Protocol ) and the system only shows the below message.

[image]

Also from last a different vendor is also trying to connect the same WINCC System , from then this problem has started .

Is there network issue ? or Deadlock Issue ? or any issue pertaining the same access .

Pls help me resolve this issue .

Pls suggest me what  to do to overcome such weird problem.

Thanks

Rajnish Vishwakarma

AndreasHeine, 2021-12-14:
this is more likely an issue in your system not with OPC UA...
ask yourself what has changed in the last weeks!

closed 2021-12-14 by AndreasHeine

swamper123, 2021-12-14:
If you got can't get access via RDP, than you got other problems. If you got access to the server, you might have to reboot it or get access on it localy to figure stuff out.
As long as there isn't any code snippet, logs or sth. else, we can't say "that might be repo-relevant".