FreeOpcUa/python-opcua #1431: How to use subscription data in other thread?

JungminHaMartin, 2021-12-30:
I subscribe some nodes and I want to use the value in other thread.
How I deal with the subscripted data?
Please show the code as below.

class SubHandler(object):

    def datachange_notification(self, node, val, data):

        dict[node] = val
        Thread_Status.node_value = val

    def event_notification(self, event):
        print("Python: New event", event)

class Thread_Status(QThread):
    def __init__(self, parent):
        super().__init__(parent)
        self.parent = parent
        self.node = []
        self.node_value = []

AndreasHeine, 2021-12-30:
in python threads have shared state -> so a dict containing the variables and values with a threadlock should do the trick!

oroulet, 2021-12-30:
or put them in threading.Queue() object., depending what fits you best, They lock for you.