FreeOpcUa/python-opcua #1442: BadTooManyPublishRequests

scch1092, 2022-02-23:
Hi all, I don't know if the error has already appeared, but I get an error when subscribing. The error appears when I subscribe again after unsubscribing. I use a PLC from Beckhoff, which runs an OPC-UA server. I guess the PLC is causing problems, but how can I work around this?

The error:
""
ServiceFault from server received while waiting for publish response
exception calling callback for <Future at 0x2622e82fb48 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\Users\Chris\AppData\Local\Programs\Python\Python37\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\Users\Chris\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\client\ua_client.py", line 493, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\Users\Chris\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\client\ua_client.py", line 93, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\Chris\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\ua\uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadTooManyPublishRequests: "The server has reached the maximum number of queued publish requests."(BadTooManyPublishRequests)
""

Furthermore: sub.delete() does not work either and only returns an error:
"ServiceFault from server received while waiting for publish response"

Python 3.7.4
opcua==0.98.13
on windows 

Thanks for the help!

AndreasHeine, 2022-02-23:
hey, indeed ServiceFault is a response from the OPC UA Server! one way to work around it would be by disconnecting and reconnect then subscribe again (forces the server to delete all subs from that client/session) another way could be to wait for one keepalive-cycle before resubscribing.

fyi:
-publish requests get stored if no notification is in the queue (till keepalive, if queue still empty!)
-plc's have only limited ressources so it might be that you can have only a few of them (check the opc ua part of the manual -> "Number of Subscriptions" or "Subscriptions per Session" ...)

scch1092, 2022-02-23:
Oh wow, the answer came quickly! I will have to find out how this is implemented on the PLC. 
I will have to try it out. Thanks!

scch1092, 2022-02-23:
> hey, indeed ServiceFault is a response from the OPC UA Server! one way to work around it would be by disconnecting and reconnect then subscribe again (forces the server to delete all subs from that [...]

AndreasHeine, 2022-03-01:
without knowing what happens before the beckhof opc ua server responds with "BadTooManyPublishRequests" its a guessinggame...
opt:
1. enable debugger and post the relevant output
or
2. make a wiresharktrace of the error to see whats going on

imaginaryBuddy, 2024-05-03:
Hi, following up on this, is there any way we can catch this server status? Because I would like to have an exception handling to timeout for a bit if I receive BadTooManyPublishRequests. Appreciate the help :)