FreeOpcUa/python-opcua #1443: Regarding Service Fault Recieved During First Time Subscription While waiting Publish - Nearly 900+ Tags in one time subscription

rajnishnationfirst, 2022-02-25:
Hi Team,

I am facing issues at start during subscribing 900+ Tags for on_datachnage_notiification event. ( here i have a python list containing all 900+ Node Ids of Tags and i am suscribing the whole list as shown below in the code .)

```
if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    # logger = logging.getLogger("KeepAlive")
    # logger.setLevel(logging.DEBUG)

    client = Client("opc.tcp://AAAAAAAAAA:4861")

    try:
        client.connect()

        objects = client.get_objects_node()
        print("Objects node is: ", objects)

        node_Id_List = []
        for iterator in range(len(tags_node_ids_list)):
            **node_Id_List**.append(client.get_node(tags_node_ids_list[iterator]))

   
        # subscribing to a variable node
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_data_change(**node_Id_List**)
        time.sleep(0.1)

        # we can also subscribe to events from server
        # sub.subscribe_events()
        # sub.unsubscribe(handle)
        # sub.delete()

        # calling a method on server
        # res = obj.call_method("2:ConditionRefresh")
        # print("method result is: ", res)

    finally:
        #client.disconnect()
        print("myvar is:")
```

I am facing the below issue while subscribing for notifications as given in the snapshot 
[image]

Pls help me with the possible solutions to overcome this problem or also it will be good if anyone can post with how can i tackle the problem.

Thanks 

Rajnish Vishwakarma

AndreasHeine, 2022-02-25:
try a greater publish interval!

rajnishnationfirst, 2022-02-25:
Thanks Andreas for the support.

I have to increase the publish interval for the highlighted code in the given the  opcua client.py file Right ?

[image]

which is passed by the code from OPCUA Client , which is currently 500 milliseconds for publishing interval.

**handler = SubHandler()
sub = client.create_subscription(500, handler)**

Andreas if you can please suggest what should be the recommended publishing interval in SECONDS or MILLISEONDS for 900 Nodes Id's or 1000 Node Id's.

It will be very good of you if you can help me for the above issue .

Thanks 

Rajnish Vishwakarma

AndreasHeine, 2022-02-25:
hey @rajnishnationfirst thats actually not so easy to predict... 900 tags can sound alot if the all change multiple times a second. another point to consider is the device from which the date came from e.g. PLC Model (Server -> ServerCapabilities -> OperationalLimits which clients should always check and adopt after connecting!) in your case you request queued data each 500ms... i am not sure which "Queuesize" you use for each "MonitoredItem" if you use a bigger Queuesize on those which change faster you can use a longer publishinterval (to be clear you would not miss any valuechange there is just a slightly longer delay in the delivery process)

[image]

in a opc ua "MonitoredItem" you can define the Samplinginterval in which the opc ua server checks the datasource for changes and on top of that you can define a so called "DeadbandFilter" for example "Analogvalues" so you will only recv a "DataChangeMessage" if the value has changed a certain amount

in a subscription the "MonitoredItem" samplinginterval derives from the publishinterval if not specified as far as i remember... and if you specify it smaller then the publishinterval you need to make the queuesize large enough to hold all changes of the "MonitoredItem" for the publishinterval period!

i hope this was helpful!? the opc ua "SubscriptionService" in combination  with the "MonitoredItemService" has a learningcurve and there are a lot of things to tweak and play around with!

regards Andreas