FreeOpcUa/python-opcua #1445: set_value() for ExtensionObject

scch1092, 2022-03-03:
Hi all, I wanted to know if anyone faced a similar problem. I have created a custom type (Struct) on my OPC-UA server. This one I can read. Unfortunately I don't know how to change values in this struct. 

`variants = client.load_type_definitions() 
ex_node = client.get_node("ns=4;s=glb.exObj")
ex_node_value = ex_node.get_value()
ex_node_datavalue = ex_node.get_data_value() 
`

For testing I created the object "exObj" from this Struct type. This contains two bool variables and one float. How can I change one or more values with "set_value()" now?
[image]

Thanks in advance! =)

AndreasHeine, 2022-03-03:
should work similar like in asyncua https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-extension-object-as-method-argument.py

scch1092, 2022-03-04:
Hi, thanks for the quick reply!
The way I saw it, I have to define a custom class, because in "ua.uaprotocol_auto." my custom class is not defined. I have now created my own class, following the example in the documentation: https://python-opcua.readthedocs.io/en/latest/_modules/opcua/ua/uaprotocol_auto.html#XVType
Unfortunately it doesn't work, I get an Error, although the data looks the same.

My program code looks like this:
`
class exStruc(FrozenClass):

    ua_types = [
        ('pEnable', 'Boolean'),
        ('pReset', 'Boolean'),
        ('pVal', 'Double'),
        ]

    def __init__(self):
        self.pEnable = False
        self.pReset = False
        self.pVal = 0.0

    def __str__(self):
        return 'exStruc(pEnable:{self.pEnable}, pReset:{self.pReset}, pVal:{self.pVal})'

    __repr__ = __str__

ex = exStruc()

ex.pEnable = True
ex.pReset = None
ex.pVal = None

dv = ua.DataValue(ua.Variant(ex, ua.VariantType.ExtensionObject))
`

[image]

In the screenshot both ua.uatypes.dataValue look the same. Also the output like ex.Value.Value is identical. Unfortunately, when I send it (.set_value()) I get the error:

File "c:\users\chris\appdata\local\programs\python\python37\lib\site-packages\opcua\ua\ua_binary.py", line 135, in pack
    return struct.pack(self.format, data)

error: required argument is not a float

Do I perhaps have a mistake in thinking?

AndreasHeine, 2022-03-04:
why you dont use the server methods for creating a own DataType? creating just a class does not tell the server what to do with it and as you already found out will not register/create anything in ua.uaprotocol_auto!

https://github.com/FreeOpcUa/python-opcua/blob/67f15551884d7f11659d52483e7b932999ca3a75/opcua/server/server.py#L443

see also the example from the examples folder:
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-create-custom-structures.py

scch1092, 2022-03-04:
My idea behind this was that the data type already exists on the server; the structure is already there. With "client.load_type_definitions()" I can load this also that with ".get_value()" the correct structure is present. Therefore I wanted to create the same object with the class to send it to the server with ".set_value()". The object should have the same structure as in ".get_value()".
I will try the suggested approach!

scch1092, 2022-03-04:
Okay, I found my mistake: The cod works, but no variables must be =None. 
My PLC does not accept this and probably the library does not work then, because this data has a NoneType. That means, all variables need a value.

```
class exStruc(FrozenClass):

    ua_types = [
        ('pEnable', 'Boolean'),
        ('pReset', 'Boolean'),
        ('pVal', 'Double'),
        ]

    def __init__(self):
        self.pEnable = False
        self.pReset = False
        self.pVal = 0.0

    def __str__(self):
        return 'exStruc(pEnable:{self.pEnable}, pReset:{self.pReset}, pVal:{self.pVal})'

    __repr__ = __str__

ex = exStruc() 

ex.pEnable = True  #example Values; !=None
ex.pReset = True  #example Values; !=None
ex.pVal = 2.2 #example Values; !=None

dv = ua.DataValue(ua.Variant(ex, ua.VariantType.ExtensionObject))

ex_node = client.get_node("ns=4;s=glb.exObj")
ex_node.set_value(dv)
```

closed 2022-03-04 by scch1092

AndreasHeine, 2022-03-05:
not sure if this is the correct way to do it... it will work till the point you have optional struct fields i guess!
if you load typedefinitions from clientside it should create the classes for you on the fly.

oroulet, 2022-03-05:
yes that approach does not seem to make sense. you should create the data type server side. Load it server side calling load_type_xxx() then the object is avalable as ua.MyObject() and you can use it to set values for example.
Same on client side. Call load_type_definition() then the type is available in ua namespace. Just use it

scch1092, 2022-03-06:
Oh wow, I knew there had to be a more elegant solution to my problem! The proposed solution makes everything immensely easier, thx! Of course the proposal works :)

```
ex_node = client.get_node("ns=4;s=glb.exObj")
client.load_type_definitions() 

ex = ua.exStruc() 

ex.pEnable = True  #example Values; !=None
ex.pReset = True  #example Values; !=None
ex.pVal = 2.2 #example Values; !=None

dv = ua.DataValue(ua.Variant(ex, ua.VariantType.ExtensionObject))
ex_node.set_value(dv)

```