FreeOpcUa/python-opcua #1446: TypeError: 'NoneType' object is not iterable raised from asyncua/common/structures.py

lovasoa, 2022-03-07:
**Describe the bug** <br />
When connecting to a custom opcua server, the reader crashes with the following stack trace

```
Traceback (most recent call last):
  File "/home/olojkine/.local/lib/python3.10/site-packages/uaclient/uaclient.py", line 128, in connect
    self.client.load_enums()
  File "/home/olojkine/.local/lib/python3.10/site-packages/asyncua/sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/olojkine/.local/lib/python3.10/site-packages/asyncua/sync.py", line 52, in post
    return futur.result()
  File "/usr/lib64/python3.10/concurrent/futures/_base.py", line 446, in result
    return self.__get_result()
  File "/usr/lib64/python3.10/concurrent/futures/_base.py", line 391, in __get_result
    raise self._exception
  File "/home/olojkine/.local/lib/python3.10/site-packages/asyncua/client/client.py", line 684, in load_enums
    return await load_enums(self)
  File "/home/olojkine/.local/lib/python3.10/site-packages/asyncua/common/structures.py", line 306, in load_enums
    c = await _get_enum_values(enum_name, server.get_node(child_desc.NodeId))
  File "/home/olojkine/.local/lib/python3.10/site-packages/asyncua/common/structures.py", line 317, in _get_enum_values
    c.fields = [EnumeratedValue(enumval.DisplayName.Text, enumval.Value) for enumval in val]
TypeError: 'NoneType' object is not iterable
``` 

**To Reproduce**<br />

Unfortunately, I cannot provide the source code for our opcua server, but it uses rust's opcua-server crate.

**Expected behavior**<br />
Browsing the tree works

**Screenshots**<br />
[image]

**Version**<br />
Python-Version: 3.10<br />
python-opcua: latest pypi version

deeTEEcee, 2022-03-09:
@lovasoa did you mean to post in https://github.com/FreeOpcUa/opcua-asyncio/issues? Your stacktrace involves asyncua not python-opcua

lovasoa, 2022-03-09:
Yes, sorry ! 

I don't have the permission to do it myself, but could an owner transfer the issue ?

[image]