FreeOpcUa/python-opcua #1452: Regarding Handling Exceptions of " Service Fault Recieved from server"

rajnishnationfirst, 2022-04-05:
Hi colleagues ,

I need a help from you all regarding handling **Service Fault Received from server while waiting for publish respons**e for freeopcua example of client .

I know we can handle the opcua errors bye the below code :

```
 except ua.UaError as error:
        print("OPCUA Error OPCUA Error Raised Past", error
```

But i need to handle the errors like - Service Fault Received from server while waiting for publish response with respect to the below function of client-minimal example.py - datachange_notification.

Pls help me to support on this as i am in frustration mode now . Pls support guys.

[image]

```

class SubHandler(object):

    def datachange_notification(self, handle, node, val):

        import logging
        logging.basicConfig(level=logging.DEBUG)

        if node is True:
            activate_Shift_Wise_Breakdown(handle)

        elif node is False:

            close_Breakdown_Automatically(handle)
```

kBashar, 2022-07-15:
@rajnishnationfirst I'm having this same issue. have you found an appropriate place to put the exception handling in the code?

oroulet, 2022-07-15:
you cannot catch service faults unfortunately. We do not know where to forward that info. Your client might do absolutely nothing at that point. so your only way to detect an issue is to pull manually the server status node (or another node)