FreeOpcUa/python-opcua #1457: User authentication has security bug

ryangu18, 2022-04-26:
**Describe the bug** <br />
User authentication has security bug, even enable server.set_security_IDs(“Username”)，But I can log still in anonymously

**To Reproduce**<br />
Steps to reproduce the behavior incl code. <br />
The code comes from here https://github.com/FreeOpcUa/python-opcua/issues/1153#issuecomment-713593224
it is Mr. AndreasHeine's demonstration code, I didn't make any changes.  <br />

**Expected behavior**<br />
Create a ua server, the security policy is Basic256Sha256_SignAndEncrypt , the authentication user/pwd is 'user1' and 'pw1' <br />

**Screenshots**<br />
[image]

This screenshot is as expected, we see the security policy, and can only authenticate with username and password, anonymous is disabled,  Select ok, then, we do not connect to this server, but right-click to open properties,
[image]
At this time I see that Anonymous has been enabled, choose anonymous,
[image]

 ok , and  connect to the server, we see that the server has been connected without entering a password.

[image]

**Version**<br />
Python-Version:  python 3.9 64bit<br />
python-opcua Version (e.g. master branch, 0.9): 0.98.13
[image]

schroeder-, 2022-04-26:
UAExpert allows to select every Security Policy even if it is not support, if you go over the Properties.

AndreasHeine, 2022-04-26:
nether the less it should not be able to connect with anonymus...

ryangu18, 2022-04-26:
Note that the security policy and authorization for this server are set in the code

server.set_security_policy([
                                    # ua.SecurityPolicyType.NoSecurity,
                                    # ua.SecurityPolicyType.Basic128Rsa15_Sign,
                                    # ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt,
                                    # ua.SecurityPolicyType.Basic256Sha256_Sign,
                                    ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt
                                ])
    policyIDs = ["Username"]

My problem was that I bypassed the password entry using the method above.
If it's not a bug, sorry for my bad submission, but how do I make my expectations valid? "Anonymous login is prohibited and must be authorized with username, password

schroeder-, 2022-04-26:
I quickly scanned through the code and I think the server always accepts ever UserToken even if it is disabled via set_security_policy. The server only provides the supported UserTokens for discovery/GetEndpoints, but on create_session this is not check.

AndreasHeine, 2022-04-26:
should be checked in activate session!
[image]

schroeder-, 2022-04-26:
Its not see:
[It     
    id_token = params.UserIdentityToken
        if isinstance(id_token, ua.UserNameIdentityToken):
            if self.user_manager.check_user_token(self, id_token) == False:
                raise utils.ServiceError(ua.StatusCodes.BadUserAccessDenied)
        self.logger.info("Activated internal session %s for user %s", self.name, self.user)](https://github.com/FreeOpcUa/python-opcua/blob/858274721766907148b527916e97446dbbf6af6d/opcua/server/internal_server.py#L348-L353)

I can provide a pr in the next days.

AndreasHeine, 2022-04-26:
@schroeder- could you check asyncua aswell probably the same there!

schroeder-, 2022-04-26:
@AndreasHeine same, I can provide the same fix there.

PR #1458 (2022-05-20): reject not supported identity tokens
prevents unsupported identity tokens to create a session.
For example reject anonymous connection if not set via set_security_IDs, currently this was possible and was a security issue.
Fixes #1457
closed 2022-05-20 by oroulet