FreeOpcUa/python-opcua #1466: How to report a security issue?

SharonBrizinov, 2022-05-23:
We would like to responsibly report on a vulnerability we found in python opcua. Where should we send our detailed report?

Additionally I would like to suggest adding a security policy to the repository to help other security researchers reach out to you properly.

Thanks!
Team82 Claroty Research
https://claroty.com/team82/

SharonBrizinov, 2022-05-23:
As both libraries share the same codebase they are both vulnerable. I will try asking in opcua-asyncio too

oroulet, 2022-05-23:
Just open a bug request and use the word security. That will work.

Also when I started that project my goal was to be able to connect to the PLC around from python. Just being able to connect was a huge sucess and security was really not my concern (It is still not since 99% of the time I work on closed systems ONE pc, ONE plc with one cable)  and I am sure python-opcua is full of security issues. Just a simple code review would point out many issues, up until a few weeeks ago, you could send queries to our server, without even opening the secure channel....
Unforunately it is also the situation for opcua-asyncio.

When this is said, if someone has the time and interest to look at code and document the main issues it would be great. Even better propose fixes and implement them ;-)

SharonBrizinov, 2022-05-23:
@oroulet thanks for replying! 

I prefer not to disclose the vulnerability in public via an open issue. Any chance we could privately send it via email?
if you prefer, you can send us an email first - https://security.claroty.com/ClarotyPGPKey

oroulet, 2022-05-24:
I am not really sure what to do. We are not a private organization. If the issue is disclosed in public there is much more chance that someone is interested and fixes than if you send it to me or a few others that may not have time to look at it.

Also if someone want to find an security issues in opcua.asyncio, he probably ust need to look at code a few minutes....

SharonBrizinov, 2022-05-26:
@oroulet I really prefer disclosing the vulnerabilities to you privately, and then if you believe they should be opened to the public - that's fine with me. Some of the vuln we found affect the cpp implementation while others the python implementations. 
https://github.com/FreeOpcUa/freeopcua
https://github.com/FreeOpcUa/opcua-asyncio
https://github.com/FreeOpcUa/python-opcua