FreeOpcUa/python-opcua #1489: Implementation of security ID check

fcarli3, 2022-10-24:
I'm using python opcua server and client and I'm testing all the possible security features about authentication and communication. As the specs says, there is no implementation about security ID check, then even if I call the method ```set_security_ID```, anonymous connections are still accepted.
Is there a way to implement security ID check on a python opcua server ? Any suggestions ?

AndreasHeine, 2022-10-24:
can you describe a little more detailed how you achieve that?

schroeder-, 2022-10-24:
Was fixed in #1458, but it think there was no new pip release, so you have to use the current master.

fcarli3, 2022-10-24:
> can you describe a little more detailed how you achieve that?

This my server:

```
from opcua import Server, ua
from opcua.server.user_manager import UserManager
import sys
from time import sleep
sys.path.insert(0, "..")

users_db = {
    'user1': 'psw1'
}

endpoint = 'opc.tcp://127.0.0.1:49580'

def parsing_parameters():
    l = len(sys.argv)
    port = 49580
    proto = "opc.tcp"
    address = ''

    # Script with no parameters
    if (l == 1):
        print('''\n	Usage:

            python3 server_usr_psw.py -a <Server_Address> -p <Server_Port>

            -a
                opcua server IP address

            -p
                opcua server port (default 49580)

            ''')
        exit()

    for i in range(1, l):
        if (sys.argv[i] == '-a' and i < l):
            server_addr = sys.argv[i+1]
        elif (sys.argv[i] == '-p' and i < l):
            port = sys.argv[i+1]
        elif ((sys.argv[i] == '--help' or sys.argv[i] == '-h') and i <= l):
            print('''\n	Usage:

                    python3 server_usr_psw.py -a <Server_Address> -p <Server_Port>

                -a
                    opcua server IP address

                -p
                    opcua server port (default 49580)

                ''')
            exit()
        else:
            param = sys.argv[i].split('-')
            if (param is not None and len(param) > 1):
                print('''\n	Wrong parameters! Use --help or -h for help\n
                    Usage:

                        python3 server_usr_psw.py -a <Server_Address> -p <Server_Port>

                    -a
                        opcua server IP address

                    -p
                        opcua server port (default 49580)

[... 211 lines cut ...]
        f"Basic256Sha256,SignAndEncrypt,{client_cert},{client_private_key},{server_cert}")
    client.set_user("user1")
    client.set_password("psw1")

    print("Connecting to server at {}".format(url))

    client.connect()

    root = client.get_root_node()
    print("Object node is: ", root)
    print('Children of root are ===> ', root.get_children())
    objects = client.get_objects_node()
    children = objects.get_children()
    print('Children of objects are ===> ', children)
    print()

    variable_browsing(objects)

    client.disconnect()

if __name__ == "__main__":
    main()

```
I tried to start the server as it is, with only ```Username``` as security ID, and then I started the client without setting username and password. I think the client shouldn't be able to connect to the server in this way, but it connects in any case and get all the server information it requires.