FreeOpcUa/python-opcua #1496: Not receiving event after subscribe

Darkmagister, 2022-12-14:
**Describe the bug** <br />
i've used the client-event example to subscribe to node to get the events, the only difference is that i didn't pass the event type when i use the subscribe_event

**To Reproduce**<br />

```
sys.path.insert(0, "..")

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import Client

from opcua.tools import uals

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """
    def event_notification(self, event):
        print("New event recived: ", event)

if __name__ == "__main__":

    client = Client("opc.tcp://192.168.0.5:8119/")
    try:
        client.connect()

        # Client has a few methods to get proxy to UA nodes that should always be in address space such as Root or Objects
        root = client.get_root_node()
        print("root node is: ", root)

        # Now getting a variable node using its browse path
        obj  = client.get_node('ns=2;s=Notification/Messages')
        print("Objects node is: ", obj)

        msclt = SubHandler()
        sub = client.create_subscription(100, msclt)
        handle = sub.subscribe_events(obj)

        embed()
        sub.unsubscribe(handle)
        sub.delete()
    finally:
        client.disconnect()
```

**Expected behavior**<br />
wiev print New event recived: ... but getting nothing

**Version**<br />
Python-Version:3.8.6<br />
python-opcua Version (e.g. master branch, 0.9): 0.98.9

AndreasHeine, 2022-12-14:
does it work with UAExpert? screenshot?

AndreasHeine, 2022-12-14:
can you turn on the logger and look if there is any hint or issue according to the event?

Darkmagister, 2022-12-14:
i don't know a lot is a software from a cnc machine but i have very little to none information, a getting it it's quite difficult as they don't give a lot of info ...

Darkmagister, 2022-12-14:
> 

> i made a few tests with your script against some demo-servers (Unified Automation Stack and node-opcua stack and it actually works...)
> 
> Unified Automation Stack (UaCppDemoServer): [image]
> 
> node-opcua (opc.tcp://opcua.umati.app:4843): [image]
> 
> Have you some additional information about the server!?

i've tried downloading the 1.7.7 cpp demo server from ua, but it doesn't work for me ...  the only event i get if the python client session created ... i've never used the server demo is there something i have to do to generate an event ??

AndreasHeine, 2022-12-14:
In the UaCpp server you need to trigger them!

or you can use a online server which has some demoevents implemented "opc.tcp://opcua.umati.app:4843"

Darkmagister, 2022-12-14:
> In the UaCpp server you need to trigger them!
> 
> [image]
> 
> or you can use a online server which has some demoevents implemented "opc.tcp://opcua.umati.app:4843"

thanks, same result here now i tried to contact them telling that i can see the event from the uademoserver but not from their and let's see if something happen, thanks a lot i will update

i've also updated the opcua lib to 0.98.13 but same result

Darkmagister, 2022-12-16:
one little update is that i've tried using another library (while i'm waiting some useful info) node-opcua and that works ok, i receive the events normally

AndreasHeine, 2022-12-16:
that would take some time to find the issue! the main problem is to reproduce the behavior...

you could try opcua-asyncio (pip install asyncua) it the current active supported version and python-opcua is deprecated...

it has also a sync wrapper with very little api changes!

Darkmagister, 2022-12-16:
> that would take some time to find the issue! the main problem is to reproduce the behavior...
> 
> you could try opcua-asyncio (pip install asyncua) it the current active supported version and python-opcua is deprecated...
> 
> it has also a sync wrapper with very little api changes! 

I tried with the other lib the Asyncio and same behavior as this one ... So if there is anything I could try to get debug info it would be awesome ...
Otherwise I have to rede the communication part with another lib ... And that would take more time :)

AndreasHeine, 2023-01-05:
@Darkmagister can you try this:

https://github.com/FreeOpcUa/opcua-asyncio/discussions/1164#discussioncomment-4599915

Darkmagister, 2023-01-05:
> @Darkmagister can you try this:
> 
> [FreeOpcUa/opcua-asyncio#1164 (comment)](https://github.com/FreeOpcUa/opcua-asyncio/discussions/1164#discussioncomment-4599915)

i tried using this: subscribe_alarms_and_conditions 
instead of subscribe_events

but still same result no event show up, i haven't set the type because i'm not sure what is the nodeid of the message type