FreeOpcUa/python-opcua #1503: Getting "BadSecurityChecksFailed" every few minutes for constant read requests

cloudatlas9, 2023-02-01:
Hi, when constantly reading values from a Siemens S7's OPC UA server using a client with encryption, I get this error every few minutes:

```
opcua.client.ua_client.Socket; CRITICAL; 2023-01-31 14:38:10,453; Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:None);
opcua.client.ua_client.Socket; ERROR  ; 2023-01-31 14:38:10,457; Protocol Error;
Traceback (most recent call last):
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/client/ua_client.py", line 101, in _run
    self._receive()
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/client/ua_client.py", line 121, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/client/ua_client.py", line 129, in _call_callback
    raise ua.UaError(
opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([199])
Exception in thread Thread-12:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/client/client.py", line 70, in run
    val = server_state.get_value()
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/common/node.py", line 155, in get_value
    result = self.get_data_value()
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/common/node.py", line 164, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/common/node.py", line 275, in get_attribute
    result = self.server.read(params)
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/client/ua_client.py", line 347, in read
    data = self._uasocket.send_request(request)
  File "/home/pi/py3_venv/lib/python3.9/site-packages/opcua/client/ua_client.py", line 83, in send_request
    data = future.result(self.timeout)
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 438, in result
    raise CancelledError()
concurrent.futures._base.CancelledError
```

Connecting works fine with this basic implementation (I provide certificate and key files)
```
def connect(self, url: str, application_uri: str = "", security_string: str = "") -> None:
        try:
            # Init OPC client
            self._ua_client = Client(url, timeout=10.0)

            # Set application uri and security string if provided
            if application_uri != "":
                self._ua_client.application_uri = application_uri
            if security_string != "":
                self._ua_client.set_security_string(security_string)

            # Connect to OPC server
            self._ua_client.connect()
```
and then an infinite loop constantly reading around 20 values from the S7's UA server using these methods:

```
ua_node = client.get_node(opcua_id)
result = ua_node.get_value()
```

The only thread I could find on the internet that seems sort of related was this stackoverflow [post](https://stackoverflow.com/questions/73567575/how-i-resolve-comunication-problem-opcua-library-with-pyinstaller). 
I use certificates created using [SecurePythonOpcUaClient](https://github.com/AndreasHeine/SecurePythonOpcUaClient/tree/master/x509v3) (using the default settings in `ssl.conf` except for `subjectAltName` and the `[ subject ]` properties), Python-Version: 3.9.2 and opcua Version 0.98.13.

Could it be that certificate files can somehow cause an unstable connection? Is there anything to tweak when creating them that could potentially fix such an issue?

AndreasHeine, 2023-02-02:
@cloudatlas9 you probably need to switch to asyncua (opcua-asyncio)!

this library is deprecated and not maintained anymore

> Could it be that certificate files can somehow cause an unstable connection? Is there anything to tweak when creating them that could potentially fix such an issue?

i don't think its an issue with the cert its more likely to have a bug while renewing channel tokens...

cloudatlas9, 2023-02-02:
Okay I was about to but wanted to check before that it's not a misconfiguration on my side.

> i don't think its an issue with the cert its more likely to have a bug while renewing channel tokens...
This is a helpful judgement as I don't have much knowledge about how the UA stack works under the hood. Thank you, will switch to asyncua then!

closed 2023-02-02 by cloudatlas9