FreeOpcUa/python-opcua #1504: Problem with set value in OPC UA S7-1200

Cykorek, 2023-02-17:
I can't write data to the S7 controller. I have the OPC server activated and using the python application I want to write data to one of the exposed variables. I am able to connect to the server and read the data but I can't write my own value. Please help. Below is the code:

from opcua import Client
import time
url = "opc.tcp://192.168.1.200:4840"
# Połączenie z serwerem OPC UA
client = Client(url)
client.connect()
print("Client connected")
adres1 = client.get_node("ns=4;i=2")
adres1.set_value(2)
data1 = adres1.get_value()
print(data1)
if data1 == 2:
{
print("hahah")
}
adres1.set_value(2)
client.disconnect()

AndreasHeine, 2023-02-17:
whats the error / debug log?

Cykorek, 2023-02-17:
"The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)
  File "C:\Users\Jarek\Desktop\Programy do OPC\.vscode\Test.py", line 19, in <module>
    adres1.set_value(20)
opcua.ua.uaerrors._auto.BadWriteNotSupported: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)

AndreasHeine, 2023-02-17:
this is an example for asyncua but the issue is the same! you need to specifie the correct variant/datavalue ... python-opcua tries to guess what the value varianttype, if you dont pass a Variant/DataValue-Class which is not always correct e.g. if you pass 200 it could be an Int16 / 32 & 64 or UInt 16 / 32 / 64 ... and opcua is strict in types!

https://github.com/AndreasHeine/opcua-tutorial/blob/8f1dc98e8f2301514aff5fd83ea31fdb35c82029/client/write.py#L36

schroeder-, 2023-02-17:
See #1159

closed 2023-02-17 by schroeder-

Cykorek, 2023-02-18:
` if`

Cykorek, 2023-02-18:
Thanks for this code but I found another solution:

<from opcua import Client
from opcua import ua

url = "opc.tcp://192.168.1.200:4840"
client = Client(url)
client.connect()
print("Client connected")

adres1 = client.get_node("ns=4;i=2")

# Nowa wartość, którą chcemy przypisać do węzła
new_value = ua.Variant(1000, ua.VariantType.Int16)

# Tworzymy obiekt DataValue z nową wartością
data_value = ua.DataValue(new_value)

# Przypisujemy nową wartość do węzła
adres1.set_value(data_value)

# Pobieramy wartość węzła, aby upewnić się, że została ustawiona poprawnie
data1 = adres1.get_value()
print(data1)

client.disconnect()>

Cykorek, 2023-02-18:
The most important things to do:
-Set variables for writing and reading in tia
- enter a good node I used to download this data UA expert but it is also in tia
-good ip address of opc server. I suggest you first check if you are connected to the server
- check the type of value. I use this commend
data1 = adres1.get_data_type()
print(data1)
data1 = adres1.get_value()
print(data1)
data1 = adres1.get_access_level()
print(data1)
data1 = adres1.get_data_type_as_variant_type()
print(data1)
data1 = adres1.get_user_access_level()

AndreasHeine, 2023-02-18:
setting up the server properly is indeed the first step 😉 understanding that python is dynamicaly typed and opcua strict the second one!