FreeOpcUa/python-opcua #1510: UA-Client reconciliate subscription not working

der-joel, 2023-03-13:
If e.g. a node monitored by a subscription is deleted and then re-created the subscription will no longer receive data-change notifications for this node.
To prevent this, im trying to detect if a subscription object at client-side is still in sync with the representation of this subscription at server-side. If not the state should be synced to server state. After that any missing nodes could be re-subscribed.

I found a method that seems to implement this behaviour: `Client.reconciliate_subscription`.
However, calling this method will raise an exception: `opcua.ua.uaerrors._auto.BadNothingToDo: "There was nothing to do because the client passed a list of operations with no elements."(BadNothingToDo)`

Looking at the source code it does not seem like the failing operation has no elements:
```python3
monitored_items = node.call_method(
            ua.uatypes.QualifiedName("GetMonitoredItems"),
            ua.Variant(subscription.subscription_id, ua.VariantType.UInt32)
        )
```

Minimal example for reproduction:
```python3
from opcua import Client
from opcua.tools import SubHandler

client = Client("opc.tcp://localhost:XXXX") # replace XXXX with port
client.connect()
subscription = client.create_subscription(100, SubHandler)
client.reconciliate_subscription(subscription)
```

Full traceback:
```python3
Traceback (most recent call last):
  File "/home/opcua/subscription_reconciliation_reproduce.py", line 7, in <module>
    client.reconciliate_subscription(subscription)
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/opcua/client/client.py", line 575, in reconciliate_subscription
    monitored_items = node.call_method(
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/opcua/common/node.py", line 685, in call_method
    return opcua.common.methods.call_method(self, methodid, *args)
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/opcua/common/methods.py", line 17, in call_method
    result = call_method_full(parent, methodid, *args)
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/opcua/common/methods.py", line 40, in call_method_full
    result = _call_method(parent.server, parent.nodeid, methodid, to_variant(*args))
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/opcua/common/methods.py", line 53, in _call_method
    res.StatusCode.check()
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/opcua/ua/uatypes.py", line 218, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNothingToDo: "There was nothing to do because the client passed a list of operations with no elements."(BadNothingToDo)
```

Im using a server created with this library for testing purposes and did not change anything besides adding some nodes. 
Looking at it with a GUI-client, `GetMonitoredItems` seems to exist on the server-node referenced in `Client.reconciliate_subscription`s source-code (path: `0:Root,0:Objects,0:Server,0:GetMonitoredItems`).

Is this caused by the server not properly implementing `GetMonitoredItems`?
If yes, is there any method to get a server-side representation of a subscription for comparison, independent of server implementation?

der-joel, 2023-03-13:
Note: The same issue occurs when using the new async version of this library

Minimal Example:
```python3
from asyncua import ua
from asyncua.sync import Client
from asyncua.tools import SubHandler

client = Client("opc.tcp://localhost:5005")
client.connect()
subscription = client.create_subscription(100, SubHandler)

node = client.get_node(
            ua.FourByteNodeId(ua.ObjectIds.Server)
        )
monitored_items = node.call_method(
    ua.uatypes.QualifiedName("GetMonitoredItems"),
    ua.Variant(subscription.aio_obj.subscription_id, ua.VariantType.UInt32)
)
print(monitored_items)
```

Traceback:
```python3
Traceback (most recent call last):
  File "/home/opcua/sub_reconciliate_async.py", line 12, in <module>
    monitored_items = node.call_method(
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/asyncua/sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/asyncua/sync.py", line 52, in post
    return futur.result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 444, in result
    return self.__get_result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/asyncua/common/node.py", line 713, in call_method
    return await call_method(self, methodid, *args)
  File "/home/opcua/project2Env/lib/python3.8/site-packages/asyncua/common/methods.py", line 19, in call_method
    result = await call_method_full(parent, methodid, *args)
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/asyncua/common/methods.py", line 43, in call_method_full
    result = await _call_method(parent.session, parent.nodeid, methodid, to_variant(*args))
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/asyncua/common/methods.py", line 65, in _call_method
    res.StatusCode.check()
  File "/home/opcua/venv/project2Env/lib/python3.8/site-packages/asyncua/ua/uatypes.py", line 328, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadNothingToDo: "No processing could be done because there was nothing to do."(BadNothingToDo)
```

der-joel, 2023-03-13:
accidentaly closed this issue, sorry

AndreasHeine, 2023-03-13:
your in the wrong repo ^^ your issue belongs to opcua-asyncio (asyncua) and not to python-opcua which is depricated!

closed 2023-03-13 by AndreasHeine

AndreasHeine, 2023-03-13:
but i assume the getmonitoreditem method is not implemented and you end up here 

https://github.com/FreeOpcUa/opcua-asyncio/blob/ff6969d2b7d93cc22c517cbb2f43ef98dc643a8c/asyncua/server/address_space.py#L577