FreeOpcUa/python-opcua #1516: python OPCUA  package installation failed in Linux ARM32V7

rowshan, 2023-04-13:
### Please read, before you post!

This is a _**BUG REPORT for issues in the existing code**_. 

If you have general questions, code handling problems or ideas, please use the:

Discussionsboard: https://github.com/FreeOpcUa/python-opcua/discussions <br/> or <br />
Gitter-Channel: https://gitter.im/FreeOpcUa/python-opcua

---

To save some time, please provide us following informations, if possible:

**Describe the bug** <br />
I am trying to create arm32v7 docker image to run on MOXA edge device. But each time I am try to build the image locally it failed. The image i used it is arm32v7/python:3.9. 

**To Reproduce**<br />
Steps to reproduce the behavior incl code. <br />
`FROM arm32v7/python:3.10-slim-buster
RUN apt-get update
RUN apt-get install -y --no-install-recommends build-essential gcc libxml2-dev libxslt-dev 
RUN apt-get install -y  python3-lxml 
RUN python3 -m venv /opt/venv
# Make sure we use the virtualenv:
ENV PATH="/opt/venv/bin:$PATH"
ENV OPCUA_FILE_LOCATION "/opt/sinewaveDB"
ENV OPCUA_SERVER_IP   "opc.tcp://192.168.1.2" 
ENV OPCUA_SERVER_PORT "4840"
RUN pip3 install setuptools-rust
RUN STATIC_DEPS=true
RUN pip3 install lxml==3.4.2  --use-pep517 
RUN pip3 install --upgrade pip
RUN pip3 install --upgrade asyncio
#RUN pip3 install cryptography     
#RUN pip3 install azure-storage-blob
RUN pip3 install opcua
RUN pip3 install -r requirements.txt

ENTRYPOINT ["/opt/venv/bin/python3","./app.py"] 
`
**Expected behavior**<br />
I am trying to run opcua client so it reads PLC datas via opcua server. Normal docker container works fine without any error arm64 docker image also wroks without problem; only arm32v71 has issue running opcua.
**Screenshots**<br />
it cannot build or install lxml package and I have the following error
[image]

**Version**<br />
Python-Version:3.10<br />
python-opcua Version latest (e.g. master branch, 0.9):

oroulet, 2023-04-13:
use opcua-asyncio. python-opcua is not maintained anymore (And yes opcua-asyncio will work because it does not rely on lxml

rowshan, 2023-04-14:
Thanks for your reply! but I also tried with opcua-asyncio it failed for cryptography installation step in arm32v7/pytrhon:3.8. here is the error log: 
[image]

Here find the dockerfile:
`FROM arm32v7/python:3.8-slim-buster
WORKDIR /sinewave

RUN pip3 install asyncua

CMD [ "python3", "client.py" ]`

schroeder-, 2023-04-14:
Adding `apk add libffi-dev` ?