FreeOpcUa/python-opcua #1530: Unable to change variable value to True

Zouzou1988, 2023-10-31:
Hello Dears, 
I can read data from a server (opc ua device) without any problem. However, i can't write any variables.
Here is my code:

        def set_opcua_data():
            try:
                node = client.get_node("ns=3; i=16847616")
                node.set_value(True)
            except Exception as er:
                print(f"*** Failure: {str(er)}")

Here is the error: 

*** Failure: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)

When i use Ua expert, i can read and write to the server!

Your help please :)

schroeder-, 2023-10-31:
some server doesn't allow setting the timestamp. you need to set it via DataValue.

```
datavalue = ua.DataValue(ua.Variant(True, ua.VariantType.Boolean))
node.set_value(datavalue)
```

Zouzou1988, 2023-10-31:
Same issue with your code:

                node = client.get_node("ns=3; i=16847616")
                data_value = ua.DataValue(ua.Variant(True, ua.VariantType.Boolean))
                node.set_value(data_value)

Error: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)

schroeder-, 2023-10-31:
Try this:

`node.set_attribute(ua.AttributeIds.Value, ua.DataValue(True))
`

Zouzou1988, 2023-10-31:
same error. sorry

schroeder-, 2023-10-31:
Are you sure you can write that value? Dod you try it via ua expert?

Zouzou1988, 2023-10-31:
it works using uaexpert.

Zouzou1988, 2023-10-31:
[image]

AndreasHeine, 2023-11-01:
why is there a blank after ; in the nodeid?

Zouzou1988, 2023-11-01:
> why is there a blank after ; in the nodeid?

Do you mean « , »in the description of nodeid on UA expert ? I don’t know

oroulet, 2023-11-01:
```
datavalue = ua.DataValue(ua.Variant(True, ua.VariantType.Boolean))
node.set_value(datavalue)
``

Zouzou1988, 2023-11-01:
> ```
> datavalue = ua.DataValue(ua.Variant(True, ua.VariantType.Boolean))
> datavalue.SouceTimestamp = None
> datavalaue.ServerTimestamp = None
> node.set_value(datavalue)
> ``
> ```

*** Failure: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)