FreeOpcUa/python-opcua #1541: How to get node value

rapid-18, 2024-04-29:
I try to aceess node value but exception **"The access level does not allow reading or subscribing to the Node."(BadNotReadable)** is reported.It's quite simple in C# but I don't know how to do it with python-opcua
the code(which is quite simple but still reports error) is below:
```python
from opcua import client as OpcUaClient
class OpcUaClass():
    def __init__(self, url:str):
        self.url = url
        self.Client = OpcUaClient.client.Client(url=self.url,timeout=1)
        self.Client.session_timeout = 100000
        try:
            self.Client.connect()
            node = self.Client.get_node("ns=2;s=/Channel/MachineAxis/aaIm[2]")
            try:
                data = node.get_value()
                print(data)
            except Exception as e:
                print(e)
            self.Client.disconnect()
        except Exception as e:
            print(e)
if __name__ == "__main__":
    client = OpcUaClass(url="opc.tcp://192.168.100.1:4840")
```

oroulet, 2024-04-29:
That is a mesage from the server telling you your users is not allowed to read that node. Maybe you log in with some users in C# or not reading the same node?

rapid-18, 2024-04-30:
Unfortunately there are no other users and the node name is just copied from C# code. By the way, is it possible to connect to the server via username and password?when I do 
```python
Client.set_user("OpcUaClient")
Client.set_password("123")
Client.connect()
```
"The session cannot be used because ActivateSession has not been called."(BadSessionNotActivated)" is reported, but I guess I must connect to the server via username and password to reach for the node value

AndreasHeine, 2024-04-30:
did you call it on a class instance or the class itself?

rapid-18, 2024-05-03:
I call it on a class instance