FreeOpcUa/python-opcua #1545: Triggering DataChangeEvent Only When Variant Value Changes

Xiaoyulalala, 2024-07-16:
I have observed that in the current implementation of Free OPC UA, datachange events are only triggered when the Variant value of a node changes. This behavior raises a few questions and concerns:

	1. Why is the datachange event only triggered by changes in the Variant value?
	2. Why not compare the entire DataValue object to trigger the datachange event, including changes in the timestamp?
	3. If I want to trigger datachange events when the timestamp of the DataValue changes, is the only solution to modify the set_attribute_value method in the Address_space?

I expected that a datachange event would be triggered when any part of the DataValue (including the timestamp) changes, not just when the Variant value changes.
Currently, the datachange event is only triggered if the Variant value changes, as shown in the code snippet:

```
if old.Value != value.Value:  # only send call callback when a value change has happened
    cbs = list(attval.datachange_callbacks.items())
```

Here is a brief description of the use case: In our application, we need to ensure that subscribers are notified not only when the value changes but also when the timestamp changes. This is important for certain monitoring and logging requirements where the timestamp itself is a critical piece of information.If applicable, here is a code snippet demonstrating the current behavior and a potential modification:

```
# Current behavior
if old.Value != value.Value:  # only send call callback when a value change has happened
    cbs = list(attval.datachange_callbacks.items())

# Proposed change to trigger on any DataValue change
if old != value:  # trigger on any change in DataValue
    cbs = list(attval.datachange_callbacks.items())
```

AndreasHeine, 2024-07-16:
https://reference.opcfoundation.org/Core/Part4/v104/docs/7.17.2

see the trigger! so you should be able to change it... while creating the monitored item!

Xiaoyulalala, 2024-07-16:
Thank you for your prompt response and assistance.
I have made the following changes to my code to set up the DataChangeTrigger to StatusValueTimestamp:

```
data_change_filter = ua.DataChangeFilter()
data_change_filter.Trigger = ua.DataChangeTrigger.StatusValueTimestamp

parameters = ua.MonitoringParameters()
parameters.ClientHandle = 1
parameters.SamplingInterval = 1000
parameters.Filter = data_change_filter
parameters.QueueSize = 1

requests = []
for node in self.nodes.values():
    request = ua.MonitoredItemCreateRequest()
    request.ItemToMonitor.NodeId = node.nodeid
    request.ItemToMonitor.AttributeId = ua.AttributeIds.Value
    request.RequestedParameters = parameters
    requests.append(request)

subscription = self.server.create_subscription(period, sub_handler)
handles = subscription.create_monitored_items(requests)
```
However, I noticed the following issue, DataChangeHandler Not Triggered on Timestamp Change: Despite setting the DataChangeTrigger to StatusValueTimestamp, the handler function is not triggered when only the timestamp changes.

Could you please review the code and let me know if there is anything I am missing or doing incorrectly? Any guidance on resolving this issue would be greatly appreciated.

python version 3.9.12
opcua version 0.98.13
Thank you for your assistance.