FreeOpcUa/python-opcua #1555: Unable to install python-opcua using apt error unable to locate package

kzawad1-ces, 2025-01-27:
### Please read, before you post!

This is a _**BUG REPORT for issues in the existing code**_. 

If you have general questions, code handling problems or ideas, please use the:

Discussionsboard: https://github.com/FreeOpcUa/python-opcua/discussions <br/> or <br />
Gitter-Channel: https://gitter.im/FreeOpcUa/python-opcua

---

To save some time, please provide us following informations, if possible:

**Describe the bug** <br />
Using the command as described in the main GitHub page to install OPCUA, but getting error:

```
ubuntu@ip-x-x-x-x:~$ sudo apt install python-opcua
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python-opcua
ubuntu@ip-x-x-x-x:~$ 

```

The output error is  "Unable to locate python-opcua".

**To Reproduce**<br />
Steps to reproduce the behavior incl code. <br />
If you post code, please use the "insert Code" (which is the "< >" symbol) format for a better readability.

1. Setup a Ubuntu server or desktop machine.
2. Run the above command from the command line.

**Expected behavior**<br />
A clear and concise description of what you expected to happen.
I expect Python to install the OPCUA library through apt command.

**Screenshots**<br />
If applicable, add screenshots to help explain your problem. <br />
You can post Screenshots directly to github, please don't use 3rd Party webpages

**Version**<br />
Python-Version:<br />
Python 3.12.3
python-opcua Version (e.g. master branch, 0.9):

AndreasHeine, 2025-01-27:
"The library is deprecated."

kzawad1-ces, 2025-01-27:
@AndreasHeine Do you have an alternative?

AndreasHeine, 2025-01-27:
@kzawad1-ces see -> https://github.com/FreeOpcUa/python-opcua/blob/master/README.md

and if you want to install python packages use pip!
https://packaging.python.org/en/latest/tutorials/installing-packages/

closed 2025-01-27 by AndreasHeine