FreeOpcUa/python-opcua #1561: ROS2 Humble (C++) Communication with Siemens PLC — Modbus TCP vs OPC UA?

kalyani2706, 2025-04-14:
Hello ROS Community,

We are currently working on a project involving ROS2 Humble (C++) and need to establish communication with a Siemens PLC. We are evaluating two commonly used industrial communication protocols:

    Modbus TCP

    OPC UA

Our goal is to build a robust and future-ready integration between the ROS2 environment and industrial PLCs, especially in the context of scalable systems and Industry 4.0 readiness.

Which protocol do you recommend for integrating ROS2 Humble (C++) with Siemens PLCs?

Has anyone here implemented either protocol in real-world industrial setups?

Are there any reliable and maintained C++ libraries or ROS2 packages (client/server) that you’d recommend for either Modbus TCP or OPC UA?

Raph-67, 2025-05-02:
Hi, siemens plcs have a native opcua server. However opcua is a bit slow for real-time application. The mains comparison criteria are : 
- Ease of integration
- Cycle time of protocol
- Process features of the protocol
- Encryption/security features (ex PUT/GET in siemens plcs are not really secures and opcua will replace them completely)

oroulet, 2025-05-02:
these 2 protocols are completely different. We use both at work for different things